riscv-non-isa/riscv-iommu #10: Make field locations more consistent, and avoid spanning 32-bit words

jhauser-us, 2022-08-08:
Please avoid splitting data structure subfields unnecessarily across 32-bit words. and try also to give them more consistent locations in the various data structures to avoid unnecessary multiplexing in the IOMMU.

Let's first consider PPNs and other addresses.  In standard RV64 PTEs (page table entries), a PPN is in bits 53:10, whereas the current document often places PPNs in bits 55:12 or in bits 43:0 within a 64-bit doubleword.  Unfortunately, each different position implies extra multiplexing in the IOMMU, up to 44 bits wide, to accomodate it.

To avoid unnecessary multiplexing, it would better to always have PPNs in the same location in every doubleword, or at least minimize the number of possible locations.  I'll leave it to the IOMMU task group to decide whether locating PPNs in bits 43:0 is really justified, but locating them in bits 55:12 is obviously a gratuitous difference.  At the very least, PPNs should always be found in either bits 53:10 or 43:0 of doublewords that an IOMMU reads from memory.

More broadly, all *physical addresses* read from memory (including guest physical addresses) should normally be shifted right by 2 bits, to be aligned with PPNs.  Besides avoiding extra multiplexing as already mentioned, this also ensures for RV32 that physical addresses are always contained within a single 32-bit word, not potentially split across two words.  (Recall that RV32 permits a 34-bit physical address space, and this includes guest physical addresses for virtual machines.)  As far as I know, the RISC-V IOMMU never needs to specify finer granularity than a 32-bit word, so the least-significant two bits of a byte-granularity address shouldn't be needed.

Furthermore, there are other fields that the current IOMMU proposal places in various locations within different data structures in memory, and again, these multiple locations likely imply added multiplexors in the IOMMU.  As I will detail below, it is possible to rearrange the formats of commands, fault records, and page-request records to more consistently place fields in the following locations within a doubleword:

    bits 63:40  - DID
    bits 59:44  - GSCID
    bits 31:12  - PID
    bits 31:12  - PSCID

Note that these locations also keep each of these fields within the bounds of a single 32-bit word.

Taking all the above into account, various in-memory data structures can be rearranged as follows to achieve greater consistency and also to avoid splitting fields across 32-bit words:

A non-leaf DDT or PDT entry:

    bits 53:10  - PPN
    bit 0       - V

The `ta` doubleword of a DC structure:

    bits 31:12  - PSCID

The `ta` doubleword of a PC structure:

    bits 31:12  - PSCID
    bits 2:0    - SUM, ENS, V

An IOTINVAL command:

    bits 125:74 - address bits 63:12
    bits 59:44  - GSCID
    bit 33      - GV
    bit 32      - PSCV
    bits 31:12  - PSCID
    bit 10      - AV
    bits 9:0    - func3, opcode

An IOFENCE command:

    bits 125:64 - address bits 63:2
    bits 63:32  - data
    bits 13:12  - PW, PR
    bit 11      - WIS
    bit 10      - AV
    bits 9:0    - func3, opcode

An IODIR command:

    bits 63:40  - DID
    bit 33      - DV
    bits 31:12  - PID
    bits 9:0    - func3, opcode

An ATS command:

    bits 127:64 - payload
    bits 63:40  - RID, DSEG
    bit 33      - DSV
    bit 32      - PV
    bits 31:12  - PID
    bits 9:0    - func3, opcode

A fault record:

    bits 255:64 - iotval2, iotval
    bits 63:40  - DID
    bits 39:32  - TTYP, PRIV, PV
    bits 31:12  - PID
    bits 11:0   - CAUSE

A page-request record:

    bits 127:64 - payload
    bits 63:40  - DID
    bits 34:32  - X, PRIV, PV
    bits 31:12  - PID

Some of the memory-mapped registers defined for an IOMMU store PPNs and addresses too, or have other fields in common with the structures above.  For these registers, I would say it would be nice if consistent locations of fields can be maintained, but here it is less important because most if not all of the registers will be implemented by real flip-flops in the IOMMU, and there may not be any actual reduction of multiplexors achieved from greater consistency of field positions.  There can still be a software advantage to not splitting fields across 32-bit register boundaries, however.

That said, my own preferences for some of these registers would be as follows:

  - The "queue base" registers, `cqb`, `fqb`, and `pqb`, have their PPN field in bits 53:10 instead of 48:5.

  - Register `tr_req_ctrl` has its DID field in bits 63:40 and PID in bits 31:12, like the structures above.  (Besides consistency, this again avoids splitting DID across a 32-bit register boundary.)  For even more consistency, PV would be in bit 32; not sure about the other minor fields.

  - Register `tr_response` has its PPN field in bits 53:10, while its other fields, fault, PBMT, and S, are moved down to bits 3:0.  (That should make the small fields slightly easier for software to mask off and examine.)

ved-rivos, 2022-08-11:
Reviewed in IOMMU TG on 8/11/2022.
TG also reviewed and approved the PR - https://github.com/riscv-non-isa/riscv-iommu/pull/14

closed 2022-08-31 by ved-rivos

yanhe234, 2024-10-23:
[image]
Hello sir, why is CQB's base address [53:10]? The bit width here differs from that of ARM, and its read address composition may be related to software drivers, but I am not very clear. Well, regardless of ARM implementation, under RV, I understand that when reading the command queue, the lower 12 bits of the address come from the lower 8 bits of CQH and the 0 bits of 4 bits, right?

ved-rivos, 2024-10-23:
The maximum physical address width supported by RISC-V is 56 bits and is larger than that supported by ARM. What is held in `cqb` register is the PPN and not the base address. The PPN * 4096 is the base address. Once the base address has been obtained the `cqh` and `cqt` act as indices into the queue.

yanhe234, 2025-03-10:
Hello, sir. I would like to ask, for pq and fq, when it exceeds 4k, will the low bit of ppn for cqb and pqb definitely default to 0?

ved-rivos, 2025-03-10:
It is software responsibility to program a PPN in `cqb` or `pqb` that satisfies the alignment requirement as determined by the `LOG2SZ-1` field. If the base address as determined by PPN is not aligned as required, all entries in the queue appear to an IOMMU as UNSPECIFIED and any address an IOMMU may compute and use for accessing an entry in the queue is also UNSPECIFIED. It is legal for an IOMMU implementation to compute the address of a queue entry as `base_address | index * size` instead of `base_address + index * size`.

yanhe234, 2025-03-10:
many  thanks sir