riscv-non-isa/riscv-iommu #9: Clarify effects of little-endian versus big-endian configuration

jhauser-us, 2022-08-08:
The effects of choosing little-endian versus big-endian in register `fctrl` need some clarification.

When an IOMMU is configured for big-endian memory accesses, what are the sizes for which bytes are reversed in memory in each instance?  For MSI PTEs, it is explicitly stated that each doubleword is reversed, but for other data structures nothing is specified.  For example, given that an extended-format device context is shown as a single entity of 512 bits, should we assume that changing the endianness mode reverses all 64 bytes in memory, as though the structure were one 512-bit integer?  When configured for little-endian, I presume the `tc` doubleword of this structure is meant as the first 8 bytes, but when configured for big-endian, should we assume the first 8 bytes (at the lowest addresses) are the reserved bytes, and `tc` is the last 8 bytes (at the highest addresses)?

I recommend instead, for non-array data structures larger than 8 bytes, that the configured endianness determines the byte order within each 64-bit doubleword in memory but that the order of doublewords within each structure is invariant of endianness.  (Except that MRIF doublewords are always little-endian, as the document already says.)  This would be most consistent with what the proposed Advanced Interrupt Architecture both requires and recommends (some of each).

Furthermore, it appears that an IOMMU's endianness configuration is intended to have no effect on accesses to the memory-mapped registers.  In fact, the byte order for these registers is never stated, although we can guess they're "normally" supposed to be little-endian.  Is it correct that an IOMMU's memory-mapped registers are *always* little-endian---even in the case that an IOMMU's endianness configuration is big-endian-only (field `capabilities`.END = 0 and `fctrl`.END = 1, meaning big-endian)?

ved-rivos, 2022-08-08:
Agree that the configured endianness should determine byte order within each 64-bit doubleword in memory and the order of doublewords in each structure is invariant of endianness. Yes, its correct that IOMMUs memory-mapped registers are always little-endian - even with the IOMMUs configuration is big-endian. As noted END bit description clarifies it applies to memory resident data structures but is silent on the memory-mapped registers.
Both items needs an update to clarify.

ved-rivos, 2022-08-10:
Addressed by PR#12 - https://github.com/riscv-non-isa/riscv-iommu/pull/12

closed 2022-08-31 by ved-rivos