riscv-non-isa/riscv-iommu #33: PCI "Routing ID" should be "Requester ID"?

jhauser-us, 2022-09-19:
Except for the Alternative Routing-ID Interpretation (ARI) option, all the PCIe documents I can lay my hands on say *RID* is *Requester ID*, not *Routing ID*.  Even the proposed IOMMU spec uses *requester ID* in one place.

The RISC-V IOMMU spec should match what the PCI SIG says for *RID*.

(Did the PCI SIG rename *RID* to *Routing ID* in more recent standards that are not readily available to the public?)

ved-rivos, 2022-09-19:
The acronym RID is an expansion of routing ID. This has been the case since PCIe Gen 3. A public reference here - [https://irp-cdn.multiscreensite.com/55f4b1f7/files/uploaded/02_01_PCI-SIG_Architecture_Overview.pdf](url) (page 9) and [https://www.nxp.com/docs/en/supporting-information/FTF-DES-N1852-PDF.pdf](url) (page 25). The one place where the specification says requester ID - paragraph after Fig 34. - is intentional and is referring to a field called "Requester ID" in the page request message. This field carries the routing ID of the requester.

jhauser-us, 2022-09-19:
I have a copy of *PCI Express Base Specification, Revision 3.0*, dated "November 10, 2010".  I based my comment on the fact that *Routing ID* doesn't appear in that document's glossary, while *Requester ID* does, defined as "the combination of a Requester's Bus Number, Device Number, and Function Number".

However, on further examination, I see you're right that the document also uses the term *Routing ID* in places.  Since that's a better term anyway, I withdraw this issue.

closed 2022-09-19 by jhauser-us