riscv-non-isa/riscv-iommu #35: For PCIe, what is a "segment number"?

jhauser-us, 2022-09-19:
For PCIe, what is likely to be a "segment number" in a device_id`?

As best as I can make out, for PCI generally, a "PCI bus segment" is identified by the bus number, which is already part of the Requester ID (RID).  That makes the end of this sentence in the introduction confusing:

    For PCIe, for example, the originating device may be identified by the unique 16-bit triplet of PCI bus number (8-bit), device number (5-bit), and function number (3-bit) (collectively known as routing identifier or RID) and optionally up to 8-bit segment number when the IOMMU supports multiple segments.

I can't find where the document explains anything about what it means for a RISC-V IOMMU to "support multiple segments", assuming this has to be something other than the usual bus number.

My guess is that by *segment* you mean what the PCIe standards call a *PCI Segment Group*, which is actually different from a PCI segment.  If that's the intention, then the end of the sentence above would be clearer as

    ... optionally up to 8-bit group number when the IOMMU supports multiple PCI segment groups.

This implies also a few other changes:

In Section 3.1.4, "IOMMU ATS commands", rename fields DSEG and DSV to something else (DGROUP and DGV?), and change the last sentence to say *group* instead of *segment*.

And in Section 6.3.7, "Device Address Translation Cache invalidation’s", change this sentence to match:

    When the IOMMU supports multiple segments then the `RID` must be qualified by the destination segment number by setting `DSV`=1 with the segment number provided in `DSEG`.

If my assumptions above are wrong, then explain in the introduction what is a PCI segment when it can't be the same as what the PCI SIG calls a PCI segment.

ved-rivos, 2022-09-19:
So PCIe topology is a tree that is rooted in the root port - a hierarchy. Each hierarchy can have up to 256 buses. Each bus can host 32 devices and each function can host 8 functions. The tuple of bus/device/function numbers (or just bus/function numbers, in ARI cases) are unique within a hierarchy. In some contexts the hierarchy is also called a segment. Before PCIe Gen 6, the segment number was not signaled part of transactions. With Gen 6, in flit mode, the segment number is also sometimes included in the ID of a function. The destination segment and destination segment valid (DSV), are two fields defined in the PCIe Gen 6 flit mode headers called OHC - orthogonal header content; specifically in OHC-A3, OHC-A4, and OHC-A5. In PCIe glossary, the term segment references to Hierarchy.

jhauser-us, 2022-09-19:
I don't have a copy of the spec for PCI Gen 6, but that's not how the PCI standards used the term *segment" up through PCI Express 3.0.  In the standards I have, what I said at the top of this issue about bus segments and segment groups is true.  The "segment number" is the bus number.  So, yes, "segment number" (bus number) has been indicated as part of the transaction long before Gen 6.

If the PCI SIG is now reusing the word *segment* to mean something confusingly different than it used to, that's bad on them.  But so far, I only have your word that they are.

I can present evidence from the earlier PCI specs as soon as I figure out how to include pictures here.

ved-rivos, 2022-09-22:
Adding following note to "Introduction" and "IOMMU ATS commands" section:

+[NOTE]
+====
+A Hierarchy is a PCI Express I/O interconnect topology, wherein the
+Configuration Space addresses, referred to as the tuple of Bus/Device/Function
+Numbers, are unique. In some contexts, a Hierarchy is also called a Segment, and
+in Flit Mode, the Segment number is sometimes included in the ID of a Function.
+====

ved-rivos, 2022-09-22:
Issue approved in IOMMU TG - 9/22/22

closed 2022-09-29 by ved-rivos