riscv-non-isa/riscv-iommu #36: Is bit DSV needed in an ATS command?

jhauser-us, 2022-09-19:
As I understand it, in normal practice for PCIe, if there is only one PCI segment group, then it is group number 0, and all PCI segments (buses) are in this group.  If that's correct, and assuming the DSEG field of an ATS command is intended as the PCI segment group number, what does it accomplish to set bit DSV = 0 that would be different from DSV = 1, when DSEG = 0?

(See also #35.)

ved-rivos, 2022-09-22:
(Please see comments in #35 and related PR #45 for definition of PCI segment number).
PCIe OHC-A4 header that is required for ID-Routed messages that require Destination Segment and/or PASID has the following fields defined:
- PASID (bits 23:16, bits 11:0), Execute-Requested (ER) (bit 14), Privilege Mode Requested (PMR) (bit 13), PASID Valid (PV) (bit 14)
- Destination Segment (bits 31:24) and Destination Segment Valid (DSV) (bit 15)
It seems certainly possible to infer that when DSEG is 0, DSV may be inferred to be 0 and thus DSEG not explicitly transmitted with the generated message. However since this is a PCIe specific command we choose to retain the PCIe defined syntax and semantics for this field. This will allow any future PCIe extensions related to semantics of DSEG and DSV to not be  incompatible with the IOMMU specification by avoiding an assumption that PCIe seems to not also make.

ved-rivos, 2022-09-22:
IOMMU TG agreed to retaining the DSV - 9/22/22