riscv-non-isa/riscv-iommu #37: Make pdtp's MODE field more consistent with all others

jhauser-us, 2022-09-19:
For no apparent reason, the `pdtp` MODE field is encoded logically backwards from the MODE of CSRs `satp` and `hgatp` (and hence of every `iohgatp`, `iosatp`, and `iovsatp`), as well as from field iommu_mode of the IOMMU's `ddtp` register.

ved-rivos, 2022-09-20:
It could be renumbered as follows:
0 - Bare, 1 - PD8, 2 - PD17, 3 - PD20.

jhauser-us, 2022-09-20:
That seems to me as good as any other encoding in increasing-size order.

(The three PD modes could also be shifted up so PD8 starts at a higher number than 1, but I couldn't think of any motivation for doing that.)

ved-rivos, 2022-09-20:
Pull request for addressing issue:
https://github.com/riscv-non-isa/riscv-iommu/pull/43

ved-rivos, 2022-09-22:
Approved by IOMMU TG - 9/22/22

closed 2022-09-29 by ved-rivos