riscv-non-isa/riscv-iommu #39: How is PSCID used in address translation algorithm?

jhauser-us, 2022-09-19:
In Section 2.4, "Process to translate an IOVA", the translation algorithm assigns a value to PSCID but does not appear actually to use it anywhere, not even for error reporting.  The purpose for this value should be explained.

ved-rivos, 2022-09-20:
So the purpose of this the PSCID and GSCID is for use to lookup and tag address translation caches. The purpose can be explained better.

ved-rivos, 2022-09-20:
Proposed text:
The process to translate an `IOVA` uses the hardware IDs (`device_id` and 
if appliicable, the `process_id`) to locate the Device Context and if applicable,
the Process Context, to determine the page tables, `PSCID`, `GSCID`, and other
control parameters that affect the address translation and protection process. 
When address translation caches are implemented, the translation process may
use the `GSCID` and `PSCID` to associate cached translations with their address 
spaces.

ved-rivos, 2022-09-22:
Addressed in pull request #44 
https://github.com/riscv-non-isa/riscv-iommu/pull/44

ved-rivos, 2022-09-22:
Approved in IOMMU TG - 9/22/22

closed 2022-09-29 by ved-rivos