riscv-non-isa/riscv-iommu #50: Maximum width of GSCID and PSCID

ved-rivos, 2022-09-24:
The architecture presently requires support for 16 bit GSCID and 20 bit PSCID.
In smaller implementation it may be desirable to support a less wider GSCID and PSCID.

Proposed update to the text:

The number of implemented bits of GSCID width, termed GSCIDLEN,
must be be same as the supported maximum device_id width. 

Bits 15:GSCIDLEN in fields of in-memory data structures and in register
fields must be set to 0 by software. Hardware implementation may 
either ignore the bits 15:GSCIDLEN or use all or parts it GSCID.

The number of implemented bits of PSCID width, termed PSCIDLEN,
shall be the large of a) the supported maximum process_id width or 
b) GSCIDLEN. 

Bits 19:PSCIDLEN in fields of in-memory data structures and in register
field must be set to 0 by software.  Hardware implementation may 
either ignore the bits 15:PSCIDLEN or use all or parts of is PSCID.

closed 2022-09-29 by ved-rivos

yanhe234, 2024-10-09:
The bit widths of GSCID and PSCID in the privileged document are 14 and 16, respectively. However, the IOMMU document specifies that the bit widths of the two are 16 and 20. Looking forward to your answer

yanhe234, 2024-10-09:
I understand that the roles of VMID and AISD in privileged documents are the same as GSCID and PSCID, both of which are used to distinguish between virtual machines and address spaces. Are there any other considerations here? Can you provide a more detailed introduction? Thank you for your response,

ved-rivos, 2024-10-10:
The key difference is that while ASID and VMID are hart local, GSCID and PSCID are global.
Please see this note from Priv. spec:
> If a hart employs an address-translation cache, that cache must appear to be private to that hart. In particular, the meaning of an ASID is local to a hart; software may choose to use the same ASID to refer to different address spaces on different harts.

> A future extension could redefine ASIDs to be global across the SEE, enabling such options as shared translation caches and hardware support for broadcast TLB shootdown. However, as OSes have evolved to significantly reduce the scope of TLB shootdowns using novel ASID-management techniques, we expect the local-ASID scheme to remain attractive for its simplicity and possibly better scalability

yanhe234, 2024-10-10:
I have learned a lot from you, and I may need to reconsider some of your answers. If you have any questions, I will ask you again. I wish you good health and a happy family. Thank you very much