riscv-non-isa/riscv-iommu #57: Don't ignore A and D bits in a PTE

jhauser-us, 2022-09-29:
If hardware A/D update isn't enabled, then an attempt to write to a page with PTE bit D = 0 should be treated as an error, thus reporting a fault, the same as an attempt to write to a page with W = 0 (no write permission).  Likewise, any attempt to access a page with A = 0 should be considered an error and report a fault.  Currently, Section 2.1.3, "Device-context fields", says under subheading "Translation control (`tc`)":
> If GADE is 0, the IOMMU ignores the A and D bits in the PTEs; the IOMMU does not update the A or D bits and does not cause any faults based on A and/or D bit being 0. ... If SADE is 0, the IOMMU ignores the A and D bits in the PTEs; the IOMMU does not update the A or D bits and does not cause any faults based on A and/or D bit being 0.

ved-rivos, 2022-09-30:
We discussed the behavior of A/D bits in IOMMU at the IOMMU overview meeting.
I took an AI to provide the background on the design in this regard. Apologies for
being late in response. 

I am noting below some of the discussion at the IOMMU TG on this topic.

As presently defined an IOMMU may optionally support hardware updating of A/D
bits PTEs when single-stage or two-stage address translation is active. When
two stage address translation is active, and the IOMMU supports hardware A/D
bit updating, the A/D updating may be selected for the VS-stage PTEs or for
G-stage PTEs, for neither, or for both. When an IOMMU does not support hardware
updating of PTE A/D bits or if hardware updating of PTE A/D bits has not been
enabled for a stage of address translation then the IOMMU behavior is to not
cause a page-fault/guest-page-fault but to ignore the 0 setting of the A or D
bits in the PTEs.

The behavior of not faulting on A and/or D bit being 0 for a access or store
to a virtual page mapped by that PTE when hardware A/D updating is not supported
or when not enabled was due to the following reasons:

1. Large buffering requirements
    Faulting on A/D bits to request software assist for setting A and/or D bits
    implies that the transactions that encountered such PTEs must be buffered in
    the IO bridge. For example, with PCIe ports operating at Gen 6 speed, the
    port may initiate memory writes at 128 GiB/s. Even if software is able to
    handle the fault interrupt and respond to the IOMMU in say 10 microseconds,
    the IOMMU would need to have buffer storage for about 1.28 MiB. This would
    also cause older transactions to be delayed leading to potentially down
    stream issues such as packet drops in the device for example as the IO bridge
    cannot release credits for these buffers. It also leads to possibility of
    misbehavior where the VS stage for example may clear A bits frequently and
    by causing the IO bridge to backup lead to undesired performance interference
    to all guests and possibly the hypervisor itself that is sharing this IO
    port.

    Selective buffering of the transactions that encountered such PTE is also
    not a feasible option due to transaction ordering rules required by IO
    protocols.  PCIe for example, requires the following ordering:
    - Read must not pass writes
    - Completions must not pass writes
    - Writes must not pass writes
    - Read, Write, Completions may pass reads
    - Completions may pass reads and other completions (if they have different
      transaction IDs).

    Completions do not invoke the IOMMU so we can ignore them for this
    discussion.

    If a write transaction encountered a PTE with A and/or D bit 0, then no
    younger writes can be allowed to be observed. Likewise younger reads must not
    pass the buffered write. Completions may also not pass older writes.

    Typically, IO bridges of high performance SoC are built to provide enough
    buffering for covering the roundtrip to memory and memory latencies are
    typically in the 10's to low 100s of nanoseconds. Requiring buffering for
    latencies of the order of software response times is quite a large burden on
    the IO bridge hardware.

2. Deadlock conditions

    Protocols like PCIe have memory ordering rules that must not be violated. We
    outlined the ordering rules in the previous paragraph. We will now illustrate
    an example where buffering IO in PCIe may lead to deadlocks.

    In this example, hart 0 makes a read from device memory. At the same instance
    there was a write generated by the device. The device now sees the read and
    generates a completion. Due to the rule that completions may not pass writes,
    the two transactions now show up in the following order at the IO bridge:
       - Transaction 0: Write
       - Transaction 1: Completion
[... 33 lines cut ...]
    supported.

Dropping the transaction that encountered the PTE with A and or D bits being 0
however will be fatal to the device. Since if a read was dropped then the
device will hang and if a write was dropped then it violates the protocol,
causes data corruption, and could cause the device and/or the driver/software
to get into bad states.

Hence the TG arrived at option 2, i.e. ignore the A/D bits, since there was
no benefit to software of reporting a fault. If software did not wish hardware
updating when hardware updating of A/D bits is supported or if software is
working with an IOMMU that does not support hardware updating the software may
always set the A and D bits in the PTEs associated with such buffers.

We also worried about temporary loss of synchronization between different parts
of the OS - for example, if the software as part of its DMA API sets the A and
D bit but the page swapper that is harvesting A bits for identifying cold pages
clears the A bits then it would cause an IO page fault leading to a device hang.
The ignoring of the A bit would not be fatal to the device or to the page
swapper. For example, if the page swapper determined after clearing the A bit
that it wishes to swap the page out, it would quickly see that the page has
been pinned for IO by the memory manager.

So due to all of these considerations the TG arrived at the current design
of not faulting when A and/or D bits are 0 on a memory access and/or store.

jhauser-us, 2022-09-30:
> 1. Large buffering requirements
>   [...]
> 2. Deadlock conditions
>   [...]
> Due to these considerations, the TG ruled out the model of requesting software assist to set A/D bits in PTEs for IO requests.  [...]

Why, Ved, are you throwing up a lot of smoke that you know has nothing to do with the request?

Nine days ago, Sept. 21, in our converstation on this same topic outside of GitHub, you wrote:
> I included these aspects as we had some discussion at the overview if a repair and resume method may be employed.  Thanks for clarifying that trap and repair model was not suggested.

In my initial comment for this GitHub issue at the top, I said the behavior should be to "report a fault", not trap and repair.  Given how we've already been over all of this, it's astounding to me that you could've forgotten in only nine days.

In our earlier conversation, I wrote:
> When hardware A/D update isn't enabled, can you explain to us why an IOMMU must *not* report a page fault for an attempted access when A = 0, or for an attempted write when D = 0?  Independent of your claim that there's no benefit to doing so, what would be the harm that we're overlooking?

Your response included, I quote:
> As to harm, I can't think of any.

If you'd like to change that answer, then please provide an explanation, not a boatload of obfuscation.

ved-rivos, 2022-09-30:
Sorry, I was not trying to throw smoke. I am sorry if you felt that way. I was trying to page in the conversation from the email thread into the issue. But got distracted with other stuff mid way.

ved-rivos, 2022-09-30:
Thanks John for filling in that part of the conversation. As part of the thread we also gathered information on other architectures and what hypervisors do:
Two competing architectures have ignored A/D bits when H/W updating is enabled. One completing architecture has an option to ignore A bit when H/W updating is disabled; but still faults on W bit being 0. Hypervisors, when HW updating of A/D bits is not available always set the A/D bits or the W bit (as applicable) to 1 and do not cause a page fault.

Proposal to address the issue:
1. A global feature control bit - fctl.fault_on_AD_bits - is defined.
2. When H/W updating of A/D bits is not enabled or not supported for a PTE the IOMMU behavior depends on the fctl.fault_on_ADD_bit control. If the fctl.fault_on_AD_bits is 1, the IOMMU will cause an appropriate page fault. If fctl.fault_on_AD_bits is 0 then the IOMMU will ignore the PTE A/D bits.

ved-rivos, 2022-09-30:
More precisely summarizing the proposed change:
If `GADE` is 1, the IOMMU updates A and D bits in G-stage PTEs atomically. If
`GADE` is 0, the IOMMU behavior is determined by `fctl.ADFD` bit. If `fctl.ADFD` 
is 0 then the IOMMU causes a guest-page-fault. If `fctrl.ADFD` is 0, the IOMMU 
ignores the A and D bits in the PTEs; the IOMMU does not update the A or D bits 
and does not cause any faults based on A and/or D bit being 0.

If `SADE` is 1, the IOMMU updates A and D bits in S/VS-stage PTEs atomically. If
`SADE` is 0, the IOMMU behavior is determined by `fctl.ADFD` bit. If `fctl.ADFD` 
is 0 then the IOMMU causes a guest-page-fault. If `fctrl.ADFD` is 0, the IOMMU 
ignores the A and D bits in the PTEs; the IOMMU does not update the A or D bits 
and does not cause any faults based on A and/or D bit being 0.

ved-rivos, 2022-09-30:
Pull request for proposed change #63

jhauser-us, 2022-09-30:
Thanks for updating your response here.

> Proposal to address the issue:
> 1. A global feature control bit - fctl.fault_on_AD_bits - is defined.

That might be acceptable, yes.  The Architecture Review Committee will want to discuss whether it agrees at its next meeting on Tuesday.