riscv-non-isa/riscv-iommu #64: Why fault an ATS Translation Request when ddtp.iommu_mode is Bare?

jhauser-us, 2022-10-03:
Maybe this is explained somewhere in the IOMMU document, but if so I couldn't find it easily.

In Section 2.4, the process to translate an IOVA says that if `ddtp.iommu_mode` is Bare, a PCIe ATS Translation Request, or a memory access attempt with a translated address, should stop and report "Transaction type disallowed" (cause = 260).

Why can't we allow PCIe ATS translation requests when `ddtp.iommu_mode` is Bare and do one of the following:  either return the virtual address back as the translated address, or set the U bit (Untranslated access only) in the Translation Completion response?

jhauser-us, 2022-10-03:
Granted, software should ideally disable ATS from a PCIe function if it knows it won't be useful.  (By clearing the Enable bit in the function's ATS Extended Capability structure.)

The question I have is, will every device driver always know when the situation calls for ATS to be disabled?

ved-rivos, 2022-10-04:
So ATS enabling/disabling is controlled by the IOMMU driver and not device drivers. Please [see](https://elixir.bootlin.com/linux/latest/C/ident/pci_enable_ats). IOMMU drivers do not enable ATS 
in devices unless ATS is enabled in the IOMMU. Please [see](https://elixir.bootlin.com/linux/latest/source/drivers/iommu/arm/arm-smmu-v3/arm-smmu-v3.c#L2298)

closed 2022-10-06 by ved-rivos