riscv-non-isa/riscv-iommu #65: What is a "message" for address translation?

jhauser-us, 2022-10-03:
Section 2.4, "Process to translate an IOVA", says:
> When the translation process reports a fault, and the request is a Untranslated request, a Translated request, or a message, the IOMMU requests the IO bridge to abort the transaction.

What is a *message* here?

Is it an ATS translation request?

If so, it would be clearer to say so.  But then, is it correct to say that the IO bridge is responsible for aborting ATS translation requests?  The figures in introduction Section 1.3, "Placement and data flow", show the ATS interface as independent of the IO bridge.

ved-rivos, 2022-10-04:
There are two messages that the IOMMU presently processes - page-request message and invalidation completion message. The "message" should be removed from this section as its not related to the process to translate an IOVA. This is left over from before the section was restructure to move Page Request handling into separate section. 

ATS translation requests are not messages but are memory-read TLPs with address type set to "Translation Request". The completion data that includes the translated address and permissions is provided by the IOMMU.

ved-rivos, 2022-10-04:
Addressed by PR #66

closed 2022-10-06 by ved-rivos