riscv-non-isa/riscv-iommu #73: Move ADFD control to device contexts

jhauser-us, 2022-10-07:
The Architecture Review Committee requests that the function of bit ADFD in register `fctl` be moved to individual device contexts (DCs).  The ARC has a couple of reasons for this request.

First, if one accepts that the IOMMU TG may be correct that there are some circumstances when faulting on incorrect A or D bit values is unwanted, it does not necessarily follow that this condition will always exist for all virtual machines simultaneously.  If A/D faults need to be disabled for a particular VM, there is currently no option to do so without forcing A/D faults to be disabled for *all* VMs.  The ARC is concerned that this is too inflexible.

Worse, having a global ADFD bit in `fctl` undermines the ability to virtualize the IOMMU for VMs.  A hypervisor should be able to give a guest OS running in a VM the illusion that it has control of an actual IOMMU; but then what should happen if the guest OS in one VM sets its virtual `fctl`.ADFD = 0 while the guest OS in a different VM sets its virtual `fctl`.ADFD = 1?  As currently designed, a hypervisor has no way to accomodate both guest OSes with only a single global ADFD bit in the real `fctl`.

Along the same lines, we suggest also that the IOMMU TG consider carefully just how certain it is that it will never be necessary to enable/disable A/D faults separately for G-stage and VS-stage page tables.  If there is only one bit in a DC for both page tables, then separating this control will be impossible, should it ever be needed.

jhauser-us, 2022-10-07:
> then separating this control will be impossible

(I mean, without a new extension to the IOMMU spec.)

ved-rivos, 2022-10-07:
>First, if one accepts that the IOMMU TG may be correct that there are some circumstances when faulting on incorrect A or D bit values is unwanted, it does not necessarily follow that this condition will always exist for all virtual machines simultaneously.

There had been no use case that required faulting on A or D bits that could be evaluated to understand the software requirements. Software presets A and D bits unless the IOMMU supported hardware updating. If functionally faulting on access were to be required for some reason the such a fault could also be created by clearing the present bit or clearing the write bit. Faults due to A or D bits would likely be indicative of a software bug, as no useful IO could occur and such faults would require resetting and re-initializing the faulting device. So it was considered unlikely to be a feature that software relies on in either the G-stage or the S/VS-stage PTEs.

Following discussion in issue #57 the conclusion was reached that there is no harm to supporting a faulting mode and a PR #63 was created to add to add the modal behavior. 

As pointed out by this issue, the modal behavior introduce a virtualization hole and is undesirable. 

PR https://github.com/riscv-non-isa/riscv-iommu/pull/74 is created to address the virtualization hole.

ved-rivos, 2022-10-08:
Some background on A/D bit faulting - architectures such as Intel and AMD did not fault on A/D bits. Later these architectures introduced support for HW updating of A/D bits. ARM has faulted on A/D bits traditionally and later introduced a feature to not fault on A bit and also introduced HW updating of A/D bits. Based on extensive discussion here faulting mode is considered more useful and the case for the ignore mode is not accepted. 

Alternative to consider for the TG is drop the ignore mode to keep architecture simple. A ignore mode may be introduced as a future extension. A PR to consider for this option is here #75 

So the two alternatives to discuss in next TG meeting:
- Introduced per device-context & per page-table ADFD controls - #74 
- Drop ADFD and only have fault on A/D bits - #75 

Please add comments/discussions on this.

pperesse, 2022-10-10:
I would prefer not to overload the device context with many various options. So I think we should postpone the decision to add A/D Fault disable option to a future revision of the IOMMU and move with option 2 to have the IOMMU always fault on A/D bits.

pperesse, 2022-10-11:
We will discuss this topic in TG but @jhauser-us can you confirm that ARC would be ok to drop this ADFD for now and revisit it in next IOMMU rev ?

ved-rivos, 2022-10-13:
Discussed in IOMMU TG on 10/13 and agreed that option 2 is preferred.
TG requesting ARC to comment if that resolves the issue.

ved-rivos, 2022-10-13:
The ADFD change was backed pending resolution of this change.

ved-rivos, 2022-10-14:
Closed with option 2 selected.

closed 2022-10-14 by ved-rivos

jhauser-us, 2022-10-18:
> @jhauser-us can you confirm that ARC would be ok to drop this ADFD for now and revisit it in next IOMMU rev ?

Yes, that is acceptable to the ARC.