riscv-non-isa/riscv-iommu #76: Encode page-table XLEN separately from MODE field

jhauser-us, 2022-10-10:
The Architecture Review Committee disapproves of using reserved or custom values of the 4-bit MODE field of RV64 CSRs `satp` and `hgatp` to indicate Sv32 or Sv32x4 in a device context's `iosatp`/`iovsatp`/`iohgatp`.  Instead, the ARC wants page-table XLEN to be handled much like endianness:  Define a GXL field in `fctl` that determines the XLEN (32 or 64 bits) for G-stage address translation in all DCs, and in each DC, have an SXL field that determines the XLEN for S/VS-stage address translation for that DC.

ved-rivos, 2022-10-11:
1. Introduce a new WARL control field - `fctl.GXL`. When `fctl.GXL` is 0, use of RV32 G-stage PT formats is supported and when is 1 the use of RV64 G-stage PT formats is supported. 
2. Introduce a new WARL control field - `DC.tc.SXL`. When `DC.tc.SXL` is 0, use of RV32 S/VS-stage PT formats is supported for the device context and when 1 the use of RV64 S/VS-stage PT formats is supported for the device context.
3. `DC.tc.SXL` must be 0 if `fctl.GXL` is 0. When `fctl.GXL` is 1 then `DC.tc.SXL` must be one of the legal values for `fctl.GXL`. 
4. When `DC.tc.SXL` is 32
      + if S/VS-stage mode is Sv32 and incoming VA > 2^32-1 then a page fault occurs. 
      + If G-stage is active and incoming GPA > 2^34-1 then a guest page fault occurs.

pperesse, 2022-10-11:
We have capabilities bits for Sv32 and Sv32x4 support. If one implementation does not support Rv32, it must have fctl.GXL set to 1 and all DC.tc.SXL to 1 as well or should these bits be ignored by HW ?
I think that most implementations will not support Rv32. Can we make it easier for such implementation?  Maybe changing the polarity of these new bits ?

ved-rivos, 2022-10-12:
Please review PR #78 to address this issue.

ved-rivos, 2022-10-13:
PR approved by TG - 10/13/2022 - with notes in PR to be resolved

closed 2022-10-13 by ved-rivos