riscv-non-isa/riscv-iommu #77: Addresses outside valid range must fault, even for Bare mode

jhauser-us, 2022-10-10:
This issue is coming from the Architecture Review Committee, and is related to #76.

When a device context specifies that the XLEN for S/VS-stage address translation is 32 bits, then an incoming virtual address must be zero in all bits above bit 31, i.e., the maximum allowed address is 2^32 - 1.  If an address is outside this range, then a page fault must be reported.  This is required even when an S/VS-stage page table pointer has MODE = Bare.

When `fctl` specifies that the XLEN for G-stage page tables is 32 bits, then any incoming guest physical address must be zero above bit 33, i.e., the maximum allowed guest physical address is 2^34 - 1.  If an address is outside this range, then a guest page fault must be reported.  This is required even if a G-stage page table pointer has MODE = Bare.

When the XLEN for page tables is 64 bits, if physical addresses from I/O devices may be wider than 64 bits, there are similar constraints:  The maximum allowed virtual address for S/VS-stage translation is 2^64 - 1, and the maximum allowed guest physical address for G-stage translation is 2^66 - 1.

Note that this makes handling of Bare mode depend on the XLEN for address translation.

ved-rivos, 2022-10-11:
Suggest the XLEN checks as follows to avoid these restrictions:
1. When fctl.GXL is 32 then DC.tc.SXL must be 32 and iohgatp.MODE must be a RV32 G-stage mode
2. When fctl.GXL is 64 then iohgatp.MODE must be a RV64 G-stage mode
3. When DC.tc.SXL is 32
    * If S/VS-stage PT is Sv32 and IOVA > 2^32-1 then a page fault  occurs
    * If G-stage is active and incoming GPA > 2^34-1 then a guest page fault occurs

jhauser-us, 2022-10-11:
I can see we made a mistake, and the limit on virtual addresses when S/VS-stage address translation mode is Bare should be 34 bits for RV32 and 66 bits for RV64, not the 32 bits and 64 bits I said before.  This is an imperfect compromise, but necessary to avoid the problems you noted.  (One could add more configuration options to choose between 32, 33, or 34 bits, or between 64, 65, or 66 bits, but that's surely a step too far.)

Concerning this:
> 2. With RV32 VMs, when Translate-to-GPA mode is enabled, [...] These Translated requests bypass the VS-stage, as these are GPA, but still undergo G-stage address translation. This would need to be treated as a special case as the XLEN check on IOVA must not be applied to such Translated requests.

Handling this correctly would amount to only a miniscule amount of extra logic in the IOMMU, if in fact any extra logic at all, so I disagree that it's a real concern.  But you may consider it moot now anyway.

jhauser-us, 2022-10-11:
> 1. When fctl.GXL is 32 then DC.tc.SXL must be 32 and iohgatp.MODE must be a RV32 G-stage mode

I forgot to mention earlier that Andrew Waterman and I believe the RV32 MODE should be encoded as follows:

    0 = Bare
    1-7  designated for custom use
    8 = Sv32 (S/VS-stage) or Sv32x4 (G-stage)
    9-15  reserved

(I haven't run this by the full ARC, but I think it's unlikely the other members would disagree.)  This is consistent with the existing 1-bit MODE field in `satp` or `hgatp` if you assume the expansion to 4-bit MODE appends 3 bits on the *right* end, not the left.

jhauser-us, 2022-10-11:
> IOVA is defined as a VA, GPA, or SPA - as determined by the context - and cannot be wider than 64 bits. As we are not aware of a standard IO protocol with an address field wider than 64-bits, the IOVA being restricted to 64 bits is not an issue.

That's true enough if you assume PCIe or some other common standard buses currently.  However, the RISC-V IOMMU may also be used with on-chip buses that could in theory have addresses wider than 66 bits, even today.  For example, I can't find anything in the AMBA AXI Standard that explicilty prohibits AWADDR and ARADDR being wider than 66 bits if somebody wants it.

If you intend for this IOMMU specification to limit *all* inbound IOVAs to a maximum width of 64 bits (or 66 bits), this needs to be stated somewhere in the document.  I searched but couldn't find such a statement currently.  Apologies if it's already there and I missed it.

I certainly agree that a 64-bit limit on address width will have zero impact on virtually everybody into the forseeable future---and maybe literally everybody forever, for all I know.  But the IOMMU document should still state this limit for correctness, not leave it assumed.

ved-rivos, 2022-10-12:
> For example, I can't find anything in the AMBA AXI Standard that explicilty prohibits AWADDR and ARADDR being wider than 66 bits if somebody wants it.

I was only referring to "standard IO protocol" such as PCIe, CXL, OpenCAPI, Gen Z, or what may be supported by fabrics like [CMN-700](https://developer.arm.com/documentation/102308/r3p0/?lang=en). I am sorry if It came across as suggesting that it can never be done.

> But the IOMMU document should still state this limit for correctness, not leave it assumed.

It is stated in the section 1.4. "Up to 57-bit virtual-address width, 56-bit system-physical-address, and 59-bit guest-physical-address width.". Please suggest improvements to the text.

ved-rivos, 2022-10-12:
Creating PR #78 to address this issue.

ved-rivos, 2022-10-13:
PR approved by TG - 10/13/2022 - with notes in PR to be resolved

closed 2022-10-13 by ved-rivos