riscv-non-isa/riscv-iommu #79: ATS reponse to page covered by MSI PTE in MRIF mode.

ved-rivos, 2022-10-13:
Presently the response is:
.. If the transaction is a PCIe ATS translation request then return a Success
   response with R, W, and U bit set to 1. See section 2.7 for further
   details on this processing.

As part of next version of ATS, the U bit usage is proposed to be deprecated.
Please see [presentation](https://docs.google.com/presentation/d/16N6K7GomsQwOyuQNKjLBMzFOP2HImsqtyo8qi2Jtroc/edit#slide=id.p1) on this to the PCI SIG ATS 1.2 TG. An alternate response is to generate a R=W=X=0 response
i.e. signal that the page is not resident. This would lead to device asking for 
the page to be made resident using a page request and the hypervisor can
page the MRIF into an IMSIC register file.

Proposed change:
.. If the transaction is a PCIe ATS translation request then return a Success
   response with R and W bits set to 0. See section 2.7 for further
   details on this processing.
.. Update the section 2.7 to match.

This change avoids the use of U bit in the architecture and this avoids a possible
incompatibility with the new revisions of the ATS protocol.

jhauser-us, 2022-10-13:
> An alternate response is to generate a R=W=X=0 response i.e. signal that the page is not resident. This would lead to device asking for the page to be made resident using a page request

If the response to a translation request indicates that the page isn't resident, does the PCIe Standard now require that a device must request the page through the PRI?  I thought devices historically were allowed to make ATS requests without ever using the PRI, considering that ATS version 1.0 didn't have PRI, and it was added with version 1.1.

ved-rivos, 2022-10-13:
Yes, that is a concern and we are working on gathering device and OS vendor inputs. This issue is a heads up. Its not decided if we would do the U deprecation in 1.2 or in a later more ambitious 2.0. Whether a backward compatibility option would be required also depends on what we gather from  the vendors.

jhauser-us, 2022-10-13:
Do any actual PCIe devices really use ATS or PRI for MSIs?  I would've thought page-request and translation steps would be useless for the pseudo-addresses that x86 systems use for MSIs, which leads me to strongly suspect that no PCIe devices are going to bother.  I'd expect them to always simply use the MSI address they were given, untranslated and without attempting first to make the supposed "page" resident.

If I'm right, it'll be only small consequence for us if the U bit is no longer supported in translation request responses.  What am I missing?

ved-rivos, 2022-10-13:
> Do any actual PCIe devices really use ATS or PRI for MSIs?

There is FUD around this. This was never forbidden and more importantly never enforced. So with all devices out there - including devices built using FPGAs - its hart to tell.

> I would've thought page-request and translation steps would be useless for the pseudo-addresses that x86 systems use for MSIs

x86 IOMMU does actually provide you a valid translation if requested to the 0xFEExxxxx

> it'll be only small consequence for us if the U bit is no longer supported in translation request responses

Yes. Its possible that we discover that is the case and it may be inconsequential.

ved-rivos, 2022-10-25:
Deferred for now.

closed 2022-10-25 by ved-rivos