riscv-non-isa/riscv-iommu #86: fctl.BE does not affect PDTs, S/VS-stage page tables

jhauser-us, 2022-10-19:
Section 5.4 on register `fctl` says this of the BE bit:
> When 0, IOMMU accesses to memory resident data structures (e.g. DDT, PDT, in-memory queues, S/VS and G stage page tables) are performed as little-endian accesses and when 1 as big-endian accesses.

This is incorrect for PDTs and S/VS-stage page tables, for which endianness is controlled by bit SBE in a device context (assuming `capabilities`.END = 1).

ved-rivos, 2022-10-19:
This text needs to be updated.
> (assuming capabilities.END = 1).

When `capabilities.END` is 0, the IOMMU supports a single endianness as determined by `fctl.BE`. If only one endianness is supported then `fctl.BE` is hardwired to 0 or 1 - the supported endianness. The `SBE` control is only allowed to assume values that are legal for `fctl.BE`. So if `capabilities.END` is 0, then `SBE` has to be configured to be identical to `fctl.BE`.

jhauser-us, 2022-10-19:
I accept PR #88 as a fix for this issue.

closed 2022-10-20 by ved-rivos