riscv-non-isa/riscv-iommu #87: ATS responses when T2GPA = 1 must consider validity of G-stage translation

jhauser-us, 2022-10-19:
When a device context has T2GPA = 1, the response to an ATS request to translate an address (a Translation Request message from a PCIe device) must take into account whether the translation from GPA to machine physical address is valid according to the G-stage page tables.  That is, even if VS-stage address translation succeeds in converting the supplied IOVA to a GPA, if G-stage translation cannot translate this GPA, then the response to the address translation request must indicate that the original IOVA is not valid.  (For PCIe, I believe this is accomplished with a Translation Completion message that has bits W = R = 0.)

When a device receives a successful response to an address translation request, the ability to then use the translated address (in this case a GPA) for memory accesses should not be prevented by any further address translation.  In particular, if an IOVA can be successfully translated into a GPA but that GPA cannot be translated into a machine physical address, the device that requested the IOVA translation needs to know, so it can issue a request for the page to be made resident (PCIe Page Request message), which can then lead to the hypervisor or host OS being informed that the guest physical page needs to be made resident for the VM.

When T2GPA = 1, if the response to an address translation request bypasses the check for a valid GPA-to-machine-physical-address translation, then the page-request steps described above won't occur, and the use of a translated GPA for a memory access may fail unexpectedly---unless a hypervisor or host OS adopts a policy of keeping all VM guest physical memory pinned in actual physical memory.  Aside from the disadvantage of keeping all of every VM's guest physical memory pinned in physical memory, I believe this workaround would also preclude using shadow G-stage page tables to support a nested hypervisor within a VM.  It's my opinion this is not the road we should choose to take.

ved-rivos, 2022-10-19:
When T2GPA is 1, the address translation does not skip any of the translation steps. However instead of returning the SPA, the GPA is returned in response to the translation. The device itself is not aware of the T2GPA regime being active and does not expect any failures on subsequent Translated Requests generated using the GPA.

ved-rivos, 2022-10-19:
Good point. Seems like its not very obvious from the document and should be more clear about this.
The reference model also illustrates this aspect.

ved-rivos, 2022-10-19:
PR #89 created to address the issue.

jhauser-us, 2022-10-19:
I accept PR #89 as a resolution for this issue.

closed 2022-10-20 by ved-rivos