riscv-non-isa/riscv-iommu #90: Don't force queue state to zero when queue becomes inactive

jhauser-us, 2022-10-24:
When `cqcsr.cqon` transitions from 1 to 0, don't automatically zero registers `cqh` and `cqt`, or fields `fence_w_ip`, `cmd_ill`, `cmd_to`, and `cqmf`.  Registers `cqh` and `cqt` aren't active when `cqon` is zero, so they don't need to be initialized.  Furthermore, the values of `cqh` and the fault bits in `cqcsr` may be needed for determining what commands were processed and their results before the queue became inactive, if only for debugging purposes.

When software changes `cqen` from 1 to 0, if it doesn't want to receive any new command-queue interrupts, it can easily change bit `cie` from 1 to 0 at the same time.

The same applies for the fault queue and the page-request queue.  When `fqcsr.cqon` transitions from 1 to 0, don't automatically zero registers `fqh` and `fqt`, or fields `fqof` and `fqmf`.  When `pqcsr.cqon` transitions from 1 to 0, don't automatically zero registers `pqh` and `pqt`, or fields `pqof` and `pqmf`.

ved-rivos, 2022-10-26:
When `fence_w_ip`, `cmd_ill`, `cmd_to`, and `cqmf` are 1, the CQ is not operational. The `cqh` does not move anymore and no more commands are processed. Hence software for debug can always read the `cqcsr`, `cqh`, `cqt` before disabling the queue and  be assured that they will not change between when it read them and when the queue was disabled.
The `fqof` and `fqmf` when set to 1 stall FQ and the `fqt` does not move any more. The `pqof` and `pqmf` when set to 1 stall PQ and the `pqt` does not move any more. Software may read the `fqcsr`  and `pqcsr` before disabling the queue.

jhauser-us, 2022-10-26:
> Software may read the `fqcsr` and `pqcsr` before disabling the queue.

What if a new record is written to the queue, or a new interrupt event occurs, in the window of time between when `fqcsr` or `pqcsr` was last read and when `fqon` or `pqon` becomes zero (which is after `fqen` or `pqen` is set to zero)?

Currently, the document says, for example,
> When the fault-queue is off, the `fqon` bit reads 0, `fqh` is set to 0, `fqt` is set to 0 and the `fqcsr` bits `fqof`, and `fqmf` are set to 0.

I might be convinced that zeroing this state is usually harmless, but what I don't understand is why it's needed.  Why is it necessary to wipe this information, which, as I said, might be useful for debugging at a minimum?

ved-rivos, 2022-10-27:
So the intent was to avoid having to  transition from off -> on with error bits preset to avoid another possible case which is not useful to software but is an additional state to verify for hardware.  We can drop the clearing on the on->off transition and move  the clearing of error bits to the off->on transition or provide freedom to the implementation by stating the status of error bits on off->on transition is UNDEFINED - they may be retained or they may be  cleared as the two allowed behaviors.

jhauser-us, 2022-10-28:
> So the intent was to avoid having to transition from off -> on with error bits preset to avoid another possible case which is not useful to software but is an additional state to verify for hardware.

That sounds reasonable, except I can't see what would make this a difficult case.  If an interrupt flag (other than `fence_w_ip`) is 1 when a queue gets enabled, the queue gets initialized but then doesn't advance on the IOMMU side until the flag is cleared.  If I understand correctly, `fence_w_ip` has no effect on command queue processing.  Where's the complexity?

In any event ...

> We can drop the clearing on the on->off transition and move the clearing of error bits to the off->on transition

If you're committed to automatically zeroing a queue's interrupt flags, that'll be preferable.  Since it's only 4 + 2 + 2 flip-flops for the three queues, I don't think it's so necessary to give implementations total freedom for optimizing the logic around those 8 flops.

The head and tail index registers for queues are dealt with already in #99.  (Those index registers add up to 192 flip-flops.)

closed 2022-11-03 by ved-rivos