riscv-non-isa/riscv-iommu #92: Consider turning the page-request queue into a general "request queue"

jhauser-us, 2022-10-25:
The RISC-V IOMMU would be a little better future-proofed if the page-request queue were known instead as the *request queue* (RQ).  We don't know what new kinds of requests to hosts the RISC-V IOMMU may need to support in the future, perhaps for future PCIe features currently unknown to us.

This future-proofing would also entail giving each *request record* a request code, just like the command opcodes for records in the command queue.  A specific code would be defined to indicate an *ATS request message*, which currently can be either a Page Request or a Stop Marker depending on payload.  Currently, the encoding of requests allows only ATS requests.

ved-rivos, 2022-10-25:
The architecture defines the fault/event queue as the generalized  queue for IOMMU generated faults and events. Since there are no events defined yet. 

The intent of the page request queue was to serve the ATS protocol that needs to be handled by the a software agent that needs to deal with page faults. This also allows an implementation that does not need to support PCIe or does not need to support the ATS protocol from needing to build this queue.

jhauser-us, 2022-10-25:
> The architecture defines the fault/event queue as the generalized queue for IOMMU generated faults and events.

If one queue can handle all, then why have ATS page requests been given a separate queue?

(And don't say it's so implementations that don't support ATS don't need to implement the extra queue, because that's nonsensical.  If there were only one queue, then *nobody* would need to implement the page request queue.  Not having to implement it can't be the motivation for adding the page request queue to the spec.)

> This also allows an implementation that does not need to support PCIe or does not need to support the ATS protocol from needing to build this queue.

That would still be the case if the queue were used for additional purposes in the future. An implementation that does not support any of the uses of the multipurpose "request queue" would not need to build it.

ved-rivos, 2022-10-25:
So did not state that one queue can handle it all. The fault and event queue is for faults and generalized events from the IOMMU - for future proofing if a event needs to be defined that needs software attention it would be defined for the fault and event queue. The intent of the page request queue was to allow it to be serviced by the page fault handling agent that would not need to further demux it. So there is already a multipurpose queue to allow generalized notification of faults and events  from the IOMMU that needs software attention.

jhauser-us, 2022-10-26:
If the page request queue is intended only for requests/events handled exclusively by the page fault handler and everything else is expected to go to the fault/event queue, then I'll withdraw my suggestion.