riscv-non-isa/riscv-iommu #93: Consider letting software initialize cqt, fqh, and pqh

jhauser-us, 2022-10-25:
The current IOMMU spec says that when a queue is enabled, its head and tail indices are both automatically set to zero.  To quote:
> Changing `cqen` from 0 to 1 sets the `cqh` and `cqt` to 0.

> Changing `fqen` from 0 to 1, resets the `fqh` and `fqt` to 0.

> Changing `pqen` from 0 to 1, resets the `pqh` and `pqt` to 0 ....

I see no reason why the IOMMU hardware needs to zero `cqt`, `fqh`, or `pqh`.  Software can write these registers, so leaving it to software to initialize them would save a little hardware.  If software fails to do so, the consequences are already well-defined.

ved-rivos, 2022-10-25:
Zeroing of `cqt`, `fqh`, and `pqh` is not a big burden for hardware but allows for hardware as well as verification simplification by avoiding registers that are hardware writable as well as software writable.

jhauser-us, 2022-10-25:
Your response contradicts itself.  Having the hardware zero `cqt`, `fqh`, and `pqh` makes the registers be "hardware writable as well as software writable", exactly what you said should be avoided.  If you really want these registers to be only software-writable, then you should accept the change proposed by this issue.

ved-rivos, 2022-10-25:
Sorry I misread you asking for the read-only registers to be made writable. Zeroing of cqt, fqh, and pqh is driving a reset to the register. These registers are warl allow only N bits for a queue of size 2^N to be writable and reset to 0 helps enforce that.

jhauser-us, 2022-10-25:
> Zeroing of cqt, fqh, and pqh is driving a reset to the register.

Reset isn't free.  Registers without reset are generally cheaper in hardware.  That's why the RISC-V ISA says most ISA registers have unspecified value after reset.

> These registers are warl allow only N bits for a queue of size 2^N to be writable and reset to 0 helps enforce that.

It does no such thing.

ved-rivos, 2022-10-25:
It does no such thing.Please see - "Only LOG2SZ-1:0 bits are writable."

ved-rivos, 2022-10-27:
If we accept the resolution to #99 then this issue becomes moot and the clearing of `cqt`,  `fqh`, and `pqh` can be removed on off->on transition.

ved-rivos, 2022-10-27:
PR #101 to address the issue.

closed 2022-11-03 by ved-rivos