riscv-non-isa/riscv-iommu #94: Enabling/disbling a queue should set busy bit until "on" status changes

jhauser-us, 2022-10-25:
This issue is probably just a request for clarification in the document.

Each queue's `csr` register has an enable bit, an "on" bit, and a busy bit.  Because the document expressly says that enabling or disabling a queue may take some time and the "on" bit indicates the end of this delay, I believe it's not sufficiently clear whether the busy bit is expected to be asserted during this delay.  Maybe the busy bit is intended only to indicate other delays.

I suggest it would be better to be more explicit that, after a write to a queue's `csr` register changes the enable bit, the busy bit is 1 until the "on" bit matches the enable bit.

For example, the text for bit `cqen` in `cqcsr` currently says:
> The command-queue may take some time to be active following setting the `cqen` to 1.  When the command queue is active, the `cqon` bit reads 1.

I suggest maybe inserting between those two sentences something like this:
> During this delay the `busy` bit is 1.

The beginning of the next paragraph might be edited to say:
> When `cqen` is changed from 1 to 0, the command queue may stay active (with `busy` asserted) until the commands already fetched ....

Similarly for the other queues, of course.

ved-rivos, 2022-10-25:
PR #97 created to address this issue.

jhauser-us, 2022-10-26:
Accepted.

closed 2022-10-26 by ved-rivos