riscv-non-isa/riscv-iommu #98: Should propagate existing queue errors when interrupts get enabled

jhauser-us, 2022-10-25:
I recommend that changing the interrupt-enable for a queue from 0 to 1 should propagate to `ipsr` any existing errors reported in the queue's `csr` register.  In particular:

If `cqcsr.cie` is changed from 0 to 1 when any of `fence_w_ip`, `cmd_ill`, `cmd_to`, or `cqmf` are set in `cqcsr`, then `ipsr.cip` becomes 1 if not already.

If `fqcsr.fie` is changed from 0 to 1 when either of `fqof` or `fqmf` is set in `fqcsr`, then `ipsr.fip` becomes 1 if not already.

If `pqcsr.pie` is changed from 0 to 1 when either of `pqof` or `pqmf` is set in `pqcsr`, then `ipsr.pip` becomes 1 if not already.

ved-rivos, 2022-10-26:
This is the specified behavior but the documentation does not seem to be clear on this. Will create additional text to make this more explicit.

ved-rivos, 2022-10-26:
Addressed by PR #100.

jhauser-us, 2022-10-26:
Maybe in the section for `ipsr`, define a *masked command-queue event x* as being the logical AND of bits `cie` and *x* from `cqcsr`, where *x* is one of `fence_w_ip`, `cmd_ill`, `cmd_to`, or `cqmf`.  Whenever any individual masked command-queue event changes in value from 0 to 1, then `ipsr.cip` becomes 1 if not already.

Similarly with *masked fault-queue events* for `ipcsr.fip`, and *masked page-request-queue events* for `ipcsr.pip`.  Bit `fip` is also set to 1 after a new fault record is added to the fault queue.  Etc.

closed 2022-11-03 by ved-rivos