riscv-non-isa/riscv-iommu #99: Status of bits outside range for software-writable queue index

jhauser-us, 2022-10-26:
If N is the value of `cqb` field `LOG2SZ-1` and N < 31, do bits 31:(N + 1) of register `cqt` always read as zeros, or may any of those bits ever read as nonzero?

If N is the value of `fqb` field `LOG2SZ-1` and N < 31, do bits 31:(N + 1) of register `fqh` always read as zeros, or may any of those bits ever read as nonzero?

If N is the value of `pqb` field `LOG2SZ-1` and N < 31, do bits 31:(N + 1) of register `pqh` always read as zeros, or may any of those bits ever read as nonzero?

ved-rivos, 2022-10-26:
When CQ/FQ/PQ is ON, the corresponding `cqt`\`fqh`\`pqh` bits 31:(N+1) always read zeros.

jhauser-us, 2022-10-26:
Rather than masking which bits of `cqt`/`fqh`/`pqh` can be *written* by software, it would be more consistent with existing RISC-V practice to mask *reads* of the registers so that bits above `LOG2SZ` always read as zeros.  Here, "reads" can also include the uses the IOMMU makes internally of these register values to determine when a queue is empty/full.  When the value of a queue's `LOG2SZ` parameter changes, the corresponding value of `cqt`/`fqh`/`pqh` can officially be documented as *unspecified*.

This way, writes by software to these registers would always set all 32 register flip-flops inside the IOMMU, but the read masking would ensure the value seen is always within the `LOG2SZ` bounds as required.

Currently, if I start with `cqb.LOG2SZ-1` = 31, I can write any 32-bit value I want into `cqt`; then if I change `cqb.LOG2SZ-1` to 2 while keeping `cqcsr.cqen` = 0, new writes to `cqt` change only the bottom 3 bits, leaving the upper 29 bits stuck with whatever value they had before.  That's not *wrong* necessarily, but it's certainly *different* than other registers of the RISC-V ISA.

Do you have some reason for choosing this write-masking scheme, instead of the more intuitive read-masking?

ved-rivos, 2022-10-26:
If I understood right the Priv arch behavior is that the value of the previous-width CSR is copied to a temporary register of the same width. For the read-only bits of the previous-width CSR, the bits at the same positions in the temporary register are set to zeros. The width of the temporary register is changed to the new width. If the new width W is narrower than the previous width, the least-significant W bits of the temporary register are retained and the more-significant bits are discarded. If the new width is wider than the previous width, the temporary register is zero-extended to the wider width. Each writable field of the new-width CSR takes the value of the bits at the same positions in the temporary register.

ved-rivos, 2022-10-26:
If I understood right the Priv arch behavior of changing CSR width is that the value of the previous-width CSR is copied to a temporary register of the same width. For the read-only bits of the previous-width CSR, the bits at the same positions in the temporary register are set to zeros. The width of the temporary register is changed to the new width. If the new width W is narrower than the previous width, the least-significant W bits of the temporary register are retained and the more-significant bits are discarded. If the new width is wider than the previous width, the temporary register is zero-extended to the wider width. Each writable field of the new-width CSR takes the value of the bits at the same positions in the temporary register.  So when the CSR is narrowed it does not retain the old bits but they are zeroed.
For other registers, whenever XLEN in any mode is set to a value less than the widest supported XLEN, all operations must ignore source operand register bits above the configured XLEN, and must sign-extend results to fill the entire widest supported XLEN in the destination register.  So if I understand right on a write from source to a destination register, the bits beyond the width are ignored - on the write - and the old bits in the destination are not preserved but overwritten with the sign extended bit. 

When a queue is enabled, which is when the width takes effect, the bits beyond the width in the index registers are zeroed. When a queue is in mission mode, e.g. for CQ `cqon` is 1, the IOMMU behavior on changing `cqb` (including the size) is unspecified (behavior may include discarding the write) and it is recommended to disable the CQ before changing cqb. So the bits beyond the width would not be stuck with old value.

So the choice here would be similar to ignoring the bits in the source that are beyond the width of the destination register. This also helped avoid additional logic on the read data path which usually is more critical than the write data path that is infrequent and less critical.

jhauser-us, 2022-10-27:
> If I understood right the Priv arch behavior of changing CSR width is that the value of the previous-width CSR is copied to a temporary register of the same width.  [...]

You're referencing the Privileged ISA document's Section 2.6, "CSR Width Modulation", which addresses the case when XLEN changes between 32 bits and 64 bits (and potentially 128 bits in the future).  The more relevant precedent for this issue is Section 2.4, "CSR Field Modulation", which begins:
> If a write to one CSR changes the set of legal values allowed for a field of a second CSR, then unless specified otherwise, the second CSR’s field immediately gets an *unspecified* value from among its new legal values.

To be consistent with similar behavior of RISC-V CSRs, changing `LOG2SZ-1` in `cqb` would change the set of legal values allowed for `cqt`, which should always be in the range 0 to 2^`LOG2SZ` - 1 inclusive.  Following the rule from Privileged ISA Section 2.4, if I change `cqb.LOG2SZ-1` from 31 to 2, and then I immediately read from `cqt`, I should see some unspecified legal value in the range 0 to 7.  I should not see any of bits 31:3 of `cqt` be nonzero, even if they were definitely nonzero before.

Likewise for `fqh` and `pqh`, of course.

(For the read-only queue index registers, you could just say they get automatically zeroed if the base address register is written, same as when the queue is changed to enabled, which I'm not challenging.)

> This also helped avoid additional logic on the read data path which usually is more critical than the write data path that is infrequent and less critical.

That is a legitimate concern.  However, I'm not yet convinced it's enough to justify odd behavior.  I'll raise the efficiency point with the Architecture Review Committee when it convenes next Tuesday.

jhauser-us, 2022-10-27:
Actually, there's an easy compromise that deals with your efficiency concern.  The IOMMU document should say that *any* write to a queue's base address register (`cqb`, `fqb`, `pqb`) when the queue is inactive (not "on") causes both the queue's two index registers to get unspecified values in the new valid range of 0 to 2^`LOG2SZ` - 1.

Under this rule, an individual implementation of the RISC-V IOMMU can choose to always zero (i.e., reset) both a queue's index registers when the base address register is written, since zero is always in the legal range.  If you make that choice in your implementation, then you can quietly get away with doing write-masking instead of read-masking, without any violation of the proposed new rules, which will now be more consistent with standard RISC-V practice.  Nobody loses.

jhauser-us, 2022-10-27:
I wrote:
> when the queue is inactive (not "on")

More precisely, that should be, neither "on" or busy.

jhauser-us, 2022-10-27:
Better, because it's clearer:  Not enabled or "on".

ved-rivos, 2022-10-27:
That would be a good way to address this.
1. IOMMU behavior of changing `cqb` when `busy` is 1 or `cqon` is 1 is `UNSPECIFIED`. The software recommended sequence to change `cqb` is to first disable the command-queue by clearing `cqen` and waiting for both `busy` and `cqon` to be 0 before changing the `cqb`. The status of bits `31:cqb.LOG2SZ` in `cqt` when `cqb` is changed is `UNSPECIFIED` .
2. IOMMU behavior of changing `fqb` when `busy` is 1 or `fqon` is 1 is `UNSPECIFIED`. The software recommended sequence to change `fqb` is to first disable the fault-queue by clearing `fqen` and waiting for both `busy` and `fqon` to be 0 before changing the `fqb`. The status of bits `31:fqb.LOG2SZ` in `fqh`  when `fqb` is changed is `UNSPECIFIED` ..
3. IOMMU behavior of changing `pqb` when `busy` is 1 or `pqon` is 1 is `UNSPECIFIED`. The software recommended sequence to change `pqb` is to first disable the page-request-queue by clearing `pqen` and waiting for both `busy` and `pqon` to be 0 before changing the `pqb`. The status of bits `31:pqb.LOG2SZ` in `pqh`  when `pqb` is changed is `UNSPECIFIED` ..

jhauser-us, 2022-10-27:
I wrote:
> You will also be required as before to zero `cqb`

`cqh`, not `cqb`.

ved-rivos, 2022-10-27:
PR #101 to address the issue.