riscv-non-isa/riscv-iommu #104: Does Svnapot capability imply tr_response reports full NAPOT region?

jhauser-us, 2022-10-31:
If the `Svnapot` bit is set in the `capabilities` register, is the response to a debug translation request in register `tr_response` required to indicate a super-page of the full NAPOT size if the virtual address was within the range of a NAPOT group of neighboring pages according to the address translation page tables?  Or may the translation request response in `tr_response` indicate a smaller super-page size or a single page (not a super-page)?

In their address translation caches, RISC-V harts that implement Svnapot are explicitly allowed to subdivide a PTE's NAPOT region into smaller power-of-two units, even as small as single pages.  Does an IOMMU have the freedom to report such a subdivided power-of-two size in response to a translation request initiated through `tr_req_ctl`?

ved-rivos, 2022-10-31:
The IOMMU has the same freedom.

jhauser-us, 2022-10-31:
Maybe a sentence or two in the document to that effect?

ved-rivos, 2022-11-01:
An IOMMU is not required to support translations that result in superpages. So will add a more general statement that reporting translation range sizes larger than 4 KiB is optional.

closed 2022-11-03 by ved-rivos