riscv-non-isa/riscv-iommu #107: All IOMMUs should support Svnapot

jhauser-us, 2022-10-31:
Given the response to #104, the Architecture Review Committee believes that all RISC-V IOMMUs should be required to implement Svnapot.  The minimal implementation cost is to interpret bits 13:10 of a 64-bit PTE differently if bit 63 (N) is one.

If all IOMMUs must implement Svnapot, the `Svnapot` bit in register `capabilities` can be reserved for another purpose.

ved-rivos, 2022-11-01:
The features that an IOMMU must support should be part of a profile specification. Such a profile compliant IOMMU must be required to support Svnapot. A more general note should be added to the capabilities register section to indicate that some of the capabilities may be required by profiles - for example, supporting Sv39 and Sv48 is supported.

ved-rivos, 2022-11-01:
PR #109 created to add the clarification.

ved-rivos, 2022-11-03:
A further clarification was added during  TG discussion - "The IOMMU must support all the virtual memory extensions that are supported by any of the harts in the implementation."

eak, 2022-11-08:
I am slowly going through the IOMMU spec and saw this item. To add my feedback, I think you're going to need to offer some justification for what appears to be needless optionality, as Svnapot would seem to be the future direction for RISC-V, and we very much want to make sure that we're not asking the software teams to support too many variants. Keeping this in might affect time to ratification whereas decreasing the optionality will make for a smoother faster ratification, in my opinion. Svnapot is something like 20 gates of logic, and I expect it in all processors going forward, given the low cost and RVA profile requirements. When there is a not a need, we should making the software's job easier.

ved-rivos, 2022-11-08:
We expect that platform specifications will remove optionality. By requiring an IOMMU to support all virtual memory extensions supported by any of the harts in the system also constrains this to say that in a profile compliant implementation where the harts are required to support Svnapot, the IOMMU is required to support Svnapot.

ved-rivos, 2022-11-10:
From 11/8/2022 AR minutes:
**IOMMU and whether Svnapot support should be mandatory or optional.**  
AR's view continues to be that the hardware cost of supporting NAPOT PTEs in at least the trivial form is tiny (~20 gates to decode NAPOT PTEs but otherwise treat them as normal 4K PTEs), whereas in general there is software cost (whether big or little) to supporting optionality (and little hassles like extension naming support for optional features).  In short, RV arch specs should support optionality where there is value to providing implementation flexibility (and the software support cost is reasonable), and conversely supporting optionality for negligible benefit is generally discouraged.

ved-rivos, 2022-11-10:
To resolve this comment the following update to the bit 14 of capabilities register description to mark it reserved. And add a note to the capabilities register section to indicate support for Svnapot extension is mandatory.

ved-rivos, 2022-11-10:
PR #118

closed 2022-11-11 by ved-rivos