riscv-non-isa/riscv-iommu #120: For the HPM event selectors, clarify what is an "event type", ...

jhauser-us, 2022-11-15:
... and how one is supposed to determine which event types do and do not support filtering by ID.

Section 5.23, "Performance-monitoring event selector", says
> When a event type that does not support filtering by IDs is programmed then the associated counter does not increment.

What exactly is an *event type*?  If they are listed in Table 16, "Standard Events list", then which of those don't support filtering by IDs?

(And if E is an eventID code that doesn't support filtering by IDs, then how would anyone ever count those events, if the counter "does not increment"?  It seems those eventID codes must all be the same as 0, *Do not count*.)

ved-rivos, 2022-11-15:
It should read "filtering by Filter ID type"

ved-rivos, 2022-11-15:
PR #126

closed 2022-11-17 by ved-rivos