riscv-non-isa/riscv-iommu #123: Reset footprint can be reduced

jhauser-us, 2022-11-15:
In Section 5.2, "Reset behavior", it is not necessary for reset to zero all of registers `cqcsr`, `fqcsr`, and `pqcsr`.

In `cqcsr`, only bits `cqen`, `cie`, `cqon`, and `busy` need to be zeroed on reset.

In `fqcsr`, only bits `fqen`, `fie`, `fqon`, and `busy` need to be zeroed on reset.

In `pqcsr`, only bits `pqen`, `pie`, `pqon`, and `busy` need to be zeroed on reset.

ved-rivos, 2022-11-15:
This was done to actually minimize the reset foot print. Reset of the error bits (8 bits) avoided needing to specify a reset of the larger MSI configuration table state. The MSI configuration table may have X after a reset if the error bits are not reset they may lead to a memory write being trigger to unknown/unexpected memory locations.

ved-rivos, 2022-11-15:
We could constrain software including random tests to not enable interrupts. 
`fctl` can also be removed from the list.
PR #129

jhauser-us, 2022-11-16:
> if the error bits are not reset they may lead to a memory write being trigger to unknown/unexpected memory locations.

How is that possible if `cqcsr.cie`, `fqcsr.fie`, and `pqcsr.pie` are all set to zeros on reset?  (See again my list above.)

jhauser-us, 2022-11-16:
See also #137.

closed 2022-11-17 by ved-rivos