riscv-non-isa/riscv-iommu #124: Must wired interrupts from IOMMU be pulses (edge-sensitive for recipient)?

jhauser-us, 2022-11-15:
Section 5.18, "Interrupt pending status register (`ipsr`)", says
> When an interrupt-pending bit in the register is set to 1 the IOMMU will not signal another interrupt from that source till software clears that interrupt-pending bit by writing 1 to clear it.

The simplest way to connect wired interrupts from the IOMMU would be for the interrupt lines simply to reflect the values of bits 3:0 of `ipsr`.  In that case, though, each interrupt signal would remain asserted for as long as the corresponding bit is set, which is not consistent with the claim that "another interrupt" is not signaled until software clears the relevant bit in `ipsr`.

I recommend that the quoted statement above be modified to apply only to MSIs from the IOMMU, not to interrupts signaled by wires.

For wire-signaled interrupts from the IOMMU, I recommend that the interrupt wire values be direct copies of bits in `ipsr`, with the explict provision that a system may logically OR any subsets of the interrupt signals together to reduce the number of interrupt lines.  However, if this suggestion is rejected, then a comment should be added to make clear that wired interrupts must be sent as pulses and must be edge-sensitive at the receiving end.

ved-rivos, 2022-11-15:
The interrupt wires are selected by the `icvec`. This allows an implementation to determine the wire used to signal each interrupt. If an implementation supports a subset of interrupt lines then the `icvec` can be used to map to one  or more causes to a wire. In case where a implementation supports a single wire it can hardwire the `icvec` to 0 and effectively signal using a single wire.

ved-rivos, 2022-11-15:
The quoted statement while correct for MSI is not right for WSI - the level vs. edge selection for wired interrupts should be done in the connected interrupt controller. Proposed to update this statement as: "The interrupt-pending bit in the register stays 1 till software clears that interrupt-pending bit by writing 1 to clear it."

pperesse, 2022-11-15:
I agree with the updated statement from Ved.

ved-rivos, 2022-11-15:
PR #128

jhauser-us, 2022-11-16:
> The interrupt wires are selected by the `icvec`. This allows an implementation to determine the wire used to signal each interrupt.

Right.  I forgot about that feature when creating this issue.

With PR #128, it appears to me the document no longer says exactly when an MSI is sent.  Unless there's text somewhere that I'm overlooking, it now implies that an MSI is sent "constantly" while some `ipsr` bit feeding an interrupt vector is 1.  But what would that mean for MSIs really?  To try to get out of the impossibility of "constantly" for MSIs, one could guess instead that an MSI is sent each time an event occurs which is specified to set a bit in `ipsr` to 1.  If that's now the intention, it should be stated, rather than left to the reader's imagination.  However, it's probably excessive to send a new MSI even when the particular `ipsr` bit was already 1.  Hence, the original statement that an interrupt is "signaled" only when an `ipsr` bit is changed from 0 to 1 appeared to say when an MSI is sent, and without so often sending redundant MSIs unnecessarily.

But one way or another, the question of exactly when the IOMMU sends an MSI will need clarifying after PR #128.

ved-rivos, 2022-11-16:
Updated PR to clarify that MSI have edge semantics.

jhauser-us, 2022-11-17:
PR accepted.

closed 2022-11-17 by ved-rivos