riscv-non-isa/riscv-iommu #136: Report all DDT data corruption faults and internal data path errors

jhauser-us, 2022-11-16:
In Section 3.2, "Fault/Event-Queue (`FQ`)", the table of fault record `CAUSE` encodings indicates which causes are reported even if `DTF` is 1.  For DDT data corruption (code 268) and Internal data path errors (272), the table currently says "No", the causes are not reported if `DTF` is 1.  These two entries should be changed to "Yes".

Notably, if an attempt to read the DDT results in an error, then DC bit `DTF` may not be accessible, or may not be reliable.  It appears this reality was already considered for other DDT faults (which are "Yes" in the table), but DDT data corruption faults were overlooked.

ved-rivos, 2022-11-16:
That is an error. 
PR #138

closed 2022-11-17 by ved-rivos

hsxyb, 2023-02-20:
Fault cause Transaction type disallowed(code 260) can be caused by wider device id than supported in section 2.3 , it is not reported if DTF is 1.
But if the device id is misconfiged and this fault is detected before iommu get a valid DC,it may be unable to locate a valid/reliable device context later,then DTF bit may not be accessible or reliable.Does iommu just report this fault in such occasion without considering DTF bit?

ved-rivos, 2023-02-20:
If the fault is of nature such that DTF cannot be determined then the effective DTF should be by default 0.

hsxyb, 2023-02-21:
> Transaction type disallowed(code 260) is currently "NO" under the "Reported if DTF is 1?" column .Do you mean in such case iommu still report this fault even when DTF is 1(the DC may be located by the legal part of device_id)?This is a liitle confusing.

ved-rivos, 2023-02-21:
Yes, transaction type disallowed is No. If the fault is caused by a condition where the device context cannot be determined then DTF is by default 0 and the fault is reported.