riscv-non-isa/riscv-iommu #140: Is PC with ENS = 0 and SUM = 1 misconfigured?

jhauser-us, 2022-11-17:
Should this combination also be listed in Section 2.2.4, "Process-context configuration checks", as causing a PC to be considered misconfigured?:  `PC.ta.ENS` = 0 and `PC.ta.SUM` = 1.

ved-rivos, 2022-11-17:
This check was not done as it does not seem to be required. When `ENS` is 0, only unprivileged transactions proceed to the first-stage page tables and `SUM` is ignored for unprivileged request.

closed 2022-11-21 by ved-rivos

yanhe234, 2024-10-23:
Hello sir, when ENS is 1, it indicates that privileged access is supported. Privilege can access privileges, but privileged cannot access users. Only when SUM is 1, privileged can access non privileged. Is this understanding correct?
Another question is, when ENS=SUM=1, if it is a privileged access and PTE's U=1, what should be the permission for IOMMU to return to IObridge through DTI protocol? Is it the R W X of the PTE found?