riscv-non-isa/riscv-iommu #142: Clarify privilege requirements for G-stage (and MSI) address translation

jhauser-us, 2022-11-17:
I fear this point isn't made clear enough in the current document:

Consistent with the RISC-V Privileged Architecture, G-stage translation involving a regular G-stage page table (not an MSI page table) succeeds only if the leaf PTE has bit U = 1, even if the inbound memory transaction or address translation request specifies supervisor privilege (for example, a PCIe transaction or ATS request has a PASID TLP Prefix with the "Privileged Mode Requested" field set to 1).

Similarly for an MSI page table, any privilege specified by the memory transaction or address translation request is always ignored.

ved-rivos, 2022-11-17:
The specification refers to the privileged specification section 8.5 -  which has the explanation for U bit checking "when checking the U bit, the current privilege mode is always taken to be U-mode". 
The MSI PTE specification is not clear on this point. Will add a to the IOMMU specification elaborating this.

ved-rivos, 2022-11-17:
PR #143

closed 2022-11-20 by ved-rivos