riscv-non-isa/riscv-iommu #144: When T2GPA = 1, report true ATS translation even for MSI pages in MRIF mode

jhauser-us, 2022-11-17:
Section 2.6, "PCIe ATS translation request handling", says 
> If the translation request has an address determined to be an MSI address using the rules defined by the Section 2.1.3.6 but the MSI PTE is configured in MRIF mode then a Success response is generated with R, W, and U bit set to 1. 

An "Untranslated access only" response (U = 1) is necessary only when the the device context has T2GPA = 0.  When T2GPA = 1, a translation can be returned with the GPA address, if first-stage translation succeeds and execute permission wasn't requested (not allowed for MSI pages).

jhauser-us, 2022-11-17:
> if first-stage translation succeeds and execute permission wasn't requested (not allowed for MSI pages).

... and the MSI PTE is valid, not misconfigured, etc.

ved-rivos, 2022-11-17:
We debated this case previously and had decided to not specially treat this case special since in MRIF mode besides translation there is an update to the MRIF. One implementation option for MRIF was for the MRIF update to be performed by the IOMMU itself similar to how it may perform A and/or D bit updates. For such an implementation treating Translation Requests with T2GPA enable would require it to have a mode to do the MRIF PTE update conditionally. The U=1 response has similar behavior as providing a translation to a GPA.

closed 2022-11-21 by ved-rivos

jhauser-us, 2022-11-22:
> For such an implementation treating Translation Requests with T2GPA enable would require it to have a mode to do the MRIF update conditionally.

You mean, one extra bit of information to be communicated to the IOMMU for suppressing the MRIF updates?  That seems pretty trivial to me.  Besides an extra flip-flop or two, is it even 10 extra logic gates?

> The R=W=U=1 response has similar behavior as providing a translation to a GPA. The device is allowed to cache such translations. So supporting this mode has no additional benefits.

You say there are no benefits, but I see two potential ones.

First, you've reported before that the PCI SIG may remove from the PCIe standard responses with U = 1.  How will a response with U = 1 be an adequate way to handle this case if such responses aren't permitted in the future?

Second, when a DC has T2GPA = 1, inbound memory transactions that are "translated" require at most a single stage of page-table lookups, whereas "untranslated" ones obviously can require many more page-table lookups for the two-stage address translation.  It's true that some of those lookups may be cached, but that depends on the IOMMU implementation, and may put more pressure on the IOMMU's ATC.  Probabilistically, it's more likely that inbound untranslated MSIs will on average be more costly time-wise than inbound translated MSIs.

As I see it, the costs of implementing the "suppress MRIF update" mode are smaller than these potential benefits.  So FWIW, I'm not convinced the TG is making the best decision here.

jhauser-us, 2022-11-22:
On the other hand, I'd easily reverse my position if we could confirm that it's rare, bordering on nonexistent, for real PCIe devices to request translations for MSI addresses.

(To hedge our bets, part of me wants the RISC-V IOMMU spec to allow either kind of response to such ATS requests.  But I suppose the downside would then be the extra work this creates for testing.)

ved-rivos, 2022-11-23:
1. ATS is typically employed by high performance devices. It is typically
   expected that such devices used with low end IOMMU will lead to suboptimal
   results. Further, MSI/MSI-X capabilities do not have a associated PASID
   configuartion. They are not virtual addresses.
2. MSI PTEs in MRIF mode impose significant additional burden on IOMMU ATC as it
   requires significant additional storage. It is unlikely an implementation
   will design IOMMU ATC to cache MRIF mode translations. This is also the
   recommendation from the AIA specification to not attempt to do that. For high
   performance usages the MRIF is likely going to be used transiently during for
   example vCPU migration. A guest that has been parked is likely not programming
   devices to generate active IO. There might be some outstanding requests or an
   incoming request that the guest needs to wake up from MRIF mode to handle but
   there is a significant overhead associated with this transition as well for
   the hypervisor to be interrupted, reload the MRIF to a physical interrupt
   file, update MSI PTEs, perform IOTLB invalidations. The MRIFs are expected to
   be used for idle or nearly idle virtual harts and the consequence of a
   interrupt landing in a MRIF is to cause that MRIF to be loaded into a physical
   interrupt file.  So likely the MSIs handled in MRIF moded are few before the
   PTE transitions out of MRIF mode.
3. T2GPA support needs a well engineered IOMMU. Using T2GPA will negate the benefits
   of ATS completely if the IOMMU implementation is not designed to be optimal.
   MSI are a small number of transactions and T2GPA is not specific to MSI
   transactions but applicable to all transactions that such devices originate.
   Without a well engineered IOMMU, there will be significant bandwidth
   overheads and latency overheads when T2GPA is used. For example, a 512B DMA
   read which may be a typical transfer size a root complex supports when used
   in T2GPA mode mode may result in a 4 level G-stage walk. In a typical system
   with 64 byte cache lines that walk leads to 256 bytes of memory reads by the
   IOMMU to complete a 512B read - a huge bandwidth overhead and even bigger increase
   in latency. So it is unlikely that an IOMMU that desires to provide high
   performance will support T2GPA but not be engineered to do that. MSI follow
   the same as they are just a fraction of the transactions.
4. I do not want to speculate on what changes may happen downstream to
   specifications that we do not control. We will need to reach to that when
   that happens. Trying to engineer for something that may happen in future is
   not a good idea because at least I dont have a crystal ball that good.
   However, usually specifications do not change in a manner that in one fell
   swoop removes compatibility with millions of deployed devices and hosts.