riscv-non-isa/riscv-iommu #152: Cross-references to priv spec

pdonahue-ventana, 2022-11-22:
There are a few places where the spec refers to, for instance, ["Section 4.3.2 of the RISC-V privileged specification"](https://github.com/riscv-non-isa/riscv-iommu/blame/main/iommu_data_structures.adoc#L961) or "Section 8.5 of the RISC-V Privileged specification".  The [latest priv spec](https://github.com/riscv/riscv-isa-manual/releases/download/draft-20221119-5234c63/riscv-privileged.pdf) (from 2 days ago) doesn't have sections with those numbers.  One option is to refer to 5.3.2 and 9.5 (the new numbers since Smrnmi was added).  However, it would be much better to refer to section names (which tend not to change) rather than numbers (which change when new privileged extensions are added to the document).

ved-rivos, 2022-11-22:
Yes, should update to section names to avoid this.

ved-rivos, 2022-11-22:
PR #159

closed 2022-11-23 by ved-rivos