riscv-non-isa/riscv-iommu #168: iohpmevt requirement

pdonahue-ventana, 2022-11-28:
The iohpmevt1-31 section has this sentence in a non-normative comment:
> A minimum of one programmable event counter besides the cycles counter is required to comply with this specification.

Can you clarify if this is a general requirement or if it is required only if capabilities.HPM=1?  I assume the latter.

Also, it seems that this sentence should be moved to the normative text.  (It could be restating something that's already in the normative text but I didn't see it.)

ved-rivos, 2022-11-28:
In the `capabilities` register section we have this normative text "When HPM is 1, the iohpmcycles and the iohpmctr1 registers must be present and be at least 32-bits wide.". This non-normative text is providing guidance on how many may be a useful number to build. We can qualify this text with if capabilities.HPM=1.

ved-rivos, 2022-11-28:
Address in PR #170

pdonahue-ventana, 2022-11-28:
OK.  Thanks.