riscv-non-isa/riscv-iommu #169: Advice given for handling interrupts can result in dropped interrupts

jhauser-us, 2022-11-28:
When interrupts are delivered as MSIs (`fctl.WSI` = 0), some specifics of the IOMMU hardware combine to necessitate that interrupt handlers take extra precautions to avoid the possibility of dropping interrupts.  The necessary precautions aren't currently included in the advice of Section 6.3.10, "Guidelines for handling interrupts from IOMMU".

The interrupt handling guidelines of this section specify as the last step:
> 6. For each source that was serviced in this process, clear the interrupt pending bit for the source by clearing the corresponding bit in `ipsr`.

But the algorithm that's given neglects the possibility that, between the time when queue processing is unblocked and this last step, another interrupt event of the same type may occur.  Unfortunately, if that happens, the new event will not trigger the sending of a new MSI because the relevant bit in `ipsr` is still one at that time.  Assuming programmers follow the advice of this section, execution will then resume after the interrupt trap with an interrupt pending at the IOMMU but no guarantee that another trap will ever be taken to handle it.

While it is possible to fix this problem only in software (and I'll speak about that as needed), my advice includes a change to the IOMMU hardware.  Specifically, I recommend changing the rules for propagating "upstream" interrupt-pending bits from `cqcsr`, `fqcsr`, and `pqcsr` into `ipsr`.

Section 5.18, "Interrupt pending status register (`ipsr`)", says that bit `cip` is set to one in `ipsr` "if `cqcsr.cie` is 1 and any of the `fence_w_ip`, `cmd_ill`, `cmd_to`, or `cqmf` bits in `cqcsr` are 1."  I recommend this be changed to say that bit `cip` is set to one whenever the value of this expression of bits of `cqcsr` changes from 0 to 1:

    cie AND (fence_w_ip OR cmd_ill OR cmd_to OR cqmf)

Likewise, I recommend bit `fip` be set to one whenever a new record is added to the fault queue or the value of this expression of bits of `fqcsr` changes from 0 to 1:

    fie AND (fqof OR fqmf)

And of course, bit `pip` is set to one whenever a new record is added to the page-request queue or the value of this expression of bits of `pqcsr` changes from 0 to 1:

    pie AND (pqof OR pqmf)

The advantage of these rule changes is to make it easier for an interrupt handler to clear an individual bit of `ipsr` early before subsequently drilling down and addressing the underlying cause of the interrupt.  Then, if any new interrupt events of the same type occur during or after interrupt handling, they will cause the bit in `ipsr` to change back from zero to one, resulting in a new MSI being sent.  This simplifies an IOMMU interrupt handler (making software more robust) because the handler won't always need to worry that such an event might occur; it can simply return after completing its business, with the assurance that a new MSI will safely trigger a new interrupt trap as needed.

Note how the new proposed rules compare to the existing rules for counter overflow interrupts.

Without the hardware change, either an interrupt handler needs extra steps to clear `cip`, `fip`, or `pip` early in `ipsr` (first zeroing `cqcsr.cie`, `fqcsr.fie`, or `pqcsr.pie` and later resetting it) or it must execute a loop to check and handle multiple interrupts that may occur.  But I won't go further into those details now, because the effort would be mostly wasted if my hardware change is adopted.

ved-rivos, 2022-11-28:
The behavior defined for `ipsr` is that the pending bit gets set if any of the contributing bits are 1. If a second source was pending, between clearing the first source and clearing the corresponding `ipsr` bit, the `ipsr` bit gets set back to 1 and the 0->1 edge of that `ipsr` bit causes an MSI to be signaled. For example, say `cqcr.cmd_to` was 1 and caused an MSI. when software reads `cqcsr` it seems `cmd_to` as 1. It then writes what it read to `cqcsr` to clear `cmd_to` but following that write and before `ipsr.cip` is cleared, `cmd_ill` may get set to 1. When `ipsr.cip` is cleared, it will go from 1->0->1 since following clearing it would get set again due to `cqcsr.cmd_ill` being still 1. On the `ipsr.cip` 0->1 edge an MSI is occurs again.

jhauser-us, 2022-11-29:
> When `ipsr.cip` is cleared, it will go from 1->0->1 since following clearing it would get set again due to `cqcsr.cmd_ill` being still 1.

That interpretation of how `cip` transitions (1 -> 0 -> 1) makes sense but wasn't at all obvious to me.  Hence, if it's supposed to be true, it needs to be called out explicitly in the text.

However, first please note that this is not a complete solution of the problem.

Say `pqcsr.pqof` was 1 and caused an MSI.  The invoked interrupt handler manages to clear some space in the page-request queue (maybe even allocating and configuring a larger buffer, too), and clears `pqof`.  This immediately opens the door for one or more new page requests to be queued, which causes `ipsr.pip` to be set to 1.  But in fact, `pip` is still 1 from before, because we're still in the interrupt handler and haven't gotten to Step 6 yet.  When `pip` is cleared in Step 6, it transitions 1 -> 0.  Then the interrupt handler ends and normal execution resumes.

At this point, there is at least one page request in the PQ and no MSI sent for it.  There is no bounce of `pip` 1 -> 0 -> 1 because the event that "set" `pip` is done and over, and---according to my reading of the IOMMU document; maybe I'm wrong---there's nothing that forces `pip` back to 1 again.

jhauser-us, 2022-11-29:
Perhaps a good solution is to change the rules for setting `fip` and `pip` in `ipsr`.  Currently, Section 5.18, "Interrupt pending status register (`ipsr`)", says for `pip`:
> The page-request-queue-interrupt-pending is set to 1 if `pqcsr.pie` is 1 and any of the `pqof` or `pqmf` bits in `pqcsr` are 1 or if a new "Page Request" message is produced in the page-request queue.

The last part could be changed to "... or if the page-request queue is not empty".

This would be far more robust than the current rule.  Same for `fip`, of course.

jhauser-us, 2022-11-29:
> The last part could be changed to "... or if the page-request queue is not empty".

Better make it also dependent on `pqcsr.pie`, too.  (And be clear that this is the case.)  Ditto for `fip` and `fqcsr.fie`, of course.

ved-rivos, 2022-11-29:
PR #174