riscv-non-isa/riscv-iommu #171: tr_req_ctl

pdonahue-ventana, 2022-11-28:
The tr_req_ctl.RWn field is defined to be:
> When set to 1 the request only needs read-only access for this translation.

What does it mean to only need read-only access?  What if the permissions are read+write?  Maybe this should be something like "When 1, the request needs read access.  When 0, the request needs write access."

Also, the name of this field seems to be unusual.  There are lots of ways to name things but I'm not aware of anything in RISC-V that has "A, not B" encoded as ABn (as opposed to just A or maybe Bn or nB).  Assuming that you don't need write+execute permission, would it be useful to combine [3:2] into a single field with 0=write, 1=execute, 2=read, 3=reserved?

A lot of the fields in this register don't specify the 0 behavior (and don't indirectly specify it like "The PID field is valid if **and only if** this field is 1").

ved-rivos, 2022-11-28:
So a translation request could be with an intent to read or an intent to write or an intent to execute. If RWn is set to 1 then the IOMMU will not cause a fault if there is no write permission. Having write permission available when not requested is okay. This interface is modeled along lines of an ATS request.

jhauser-us, 2022-11-29:
I think my preference would be for this field's polarity to be reversed, and called *W* (Write access).  If W = 0, the request needs only read access for the translation, not write access; while if W = 1, the request needs write access.

But if the polarity isn't flipped, then I suggest the name *NW* (No Write), as that's both easily understood and is the name given for a bit with the same meaning in a PCIe ATS Translation Request.

(When I reviewed this before, I just assumed that *RWn* must be the name used by PCIe ATS, and didn't bother to actually look it up until now.)

ved-rivos, 2022-11-29:
The name for this field was debated much and went between `RWn` and `NW`. I would be okay with `NW` - we will discuss again in the TG.

ved-rivos, 2022-11-29:
PR #175

closed 2022-12-01 by ved-rivos