riscv-non-isa/riscv-iommu #173: Software steps suggested for clearing a PTE's D bit are inadequate

jhauser-us, 2022-11-29:
Section 6.3.6, "Accessed (A)/Dirty (D) bit updates and page promotions", says:
> When IOMMU supports hardware-managed A and D bit updates, if software clears the A and/or D bit in the S/VS-stage and/or G-stage PTEs then software must invalidate corresponding PTE entries that may be cached by the IOMMU.  If such invalidations are not performed, then the IOMMU may not set these bits when processing subsequent transactions that use such entries.

This is insufficient advice when the D bits in PTEs are actually being used to track dirty pages.  If a page has D = 0, a write *must not* be allowed to occur to that page without the D bit first being set.  If for some reason a write proceeds for a page while its D bit is zero, let's call that a *sneak write*.  (If there's a standard term for this, I obviously don't know it.)

The quoted sentence speaks of software clearing the D bit for a page, and also invalidating the corresponding PTEs for that page at the IOMMU, but in fact there is no ordering of those two actions alone that can provide guaranteed protection against sneak writes.

Let's assume a PTE first has D = 1, and it is cached in that form by the IOMMU.  If software takes these steps:

. Clear the PTE's D bit to zero;

. Invalidate all cached copies of the PTE at the IOMMU,

we cannot rule out that between actions 1 and 2 a write to this page may be translated and authorized by the IOMMU using its earlier cached copy.  Any such write will be what I'm calling a sneak write, and for all we can be sure, the entire page might be changed by sneak writes during the window of time between the two software actions.  The result will be a memory page with dirty content but with D = 0 in its PTE (in the main copy in memory).

The steps mentioned in the quoted text are not exactly wrong, but they are entirely inadequate without other steps in addition.  If I'm not mistaken, I believe it's necessary to

. Mark the PTE as invalid (V = 0);

. Invalidate all cached copies of the PTE at the IOMMU;

. In the PTE, set D = 0 and V = 1.

I can't see a safe sequence that does not involve invalidating the PTE for a time.

ved-rivos, 2022-11-29:
The guidance stands on its own about the behavior expected and does not make assumptions about the usage.

For example, a pre-copy VM migration phase using PTE D bit could be as follows
While not migrated
   1. For each PTE
         1. If D bit is 0 continue
         2. Clear D bit to 0
         3. Invalidate IOATC/DevATC
         4. Copy page contents to remote machine

Between 2 and 4 there may be more stores to the page but that is okay. Step 4 will copy out all modified data since last time step 4 was invoked. During/after step 4 if there are more writes then they will set the D bit again and this will be found in the next scan.

closed 2022-12-01 by ved-rivos

zetalog, 2024-02-23:
I wonder why D-bit has anything to do w/ IOMMU.
Say, D is used by OS to flag a buffer (disk cache) as dirty and needs to be flushed when necessary.
While when IOMMU sees a write targeting a D=0|1 bit page, this determination should have already been done by software (OS bdflush or similar). Modifications done by IOMMU to D-bit only triggers unexpected software behavior:
1. make a D=1 only triggers a flush a clean page back to IO?, or 
2. raise an exception on D=0 write to let software see a spurious interrupt?

For D-bit related to PTE pages, I think since ptw interface should be coherency or non-coherency depending on the SoC configuration, it should have nothing to do with de usage model of D-bit, and thus also have nothing to do with IOMMU.

ved-rivos, 2024-02-24:
@zetalog You are right that when the OS uses pinned memory for DMA it would set the A and D bits by itself. However, there are two emerging use models a) shared virtual addressing [SVA](https://docs.kernel.org/next/x86/sva.html) b) migration of VMs with direct assigned devices that uses techniques like [IOMMUFD dirty tracking](https://lore.kernel.org/all/2d369e58-8ac0-f263-7b94-fe73917782e1@linux.intel.com/T/). With SVA, a common virtual address space is shared between the host and devices and allows applications to submit work directly to devices from user space without incurring cost of pinning/unpinning memory. You are also right that an IOMMU that does not use ptw coherency will not advertise support for the either a) hardware A/D bit updating b) atomic updating of MRIF for MSI redirection to virtual machines.