riscv-non-isa/riscv-iommu #197: Why need to confirm reset value of busy bits?

jhauser-us, 2023-02-07:
Section 5.2, "Reset behavior", says that the `busy` bit in each of `cqcsr`, `fqcsr`, and `pqcsr` is reset to zero.  Given that fact, why does Section 6.2, "Guidelines for initialization", include these steps?:
> Poll on `cqcsr.busy` until it reads 0
> Poll on `fqcsr.busy` until it reads 0
> Poll on `pqcsr.busy` until it reads 0

In what circumstances would these `busy` bits not already be zero after reset?

If the expectation is that the IOMMU may have already been in use, then I believe the initialization steps have forgotten to first disable the queues by zeroing `cqcsr.cqen`, `fqcsr.fqen`, and `pqcsr.pqen`.  (But then I suspect there may be many other steps missing to safely re-init an IOMMU that was previously in-use.)

ved-rivos, 2023-02-07:
There should not be a case for these to be not 0 after a reset. The intent here was to show a safe sequence to do this initialization. For example, if software had disabled the IOMMU it may clear the `cqen`/`fqen`/`pqen` but not wait for the `busy` bit to be 0 expecting that accessor functions for `cqcsr`\`fqcsr`\`pqcsr` check the busy bit before doing a write for subsequently re-enabling the IOMMU. But seems like this is confusing. So will drop these lines.

ved-rivos, 2023-02-07:
PR #198 to drop the poll to avoid the confusion that these bits may not be 0 on reset.

closed 2023-02-09 by ved-rivos