riscv-non-isa/riscv-iommu #199: How to know that IOMMU auto-responded to a Page Request?

jhauser-us, 2023-02-07:
In Section 6.5, "Guidelines for handling interrupts from IOMMU", step 4.g includes a note:
> If a PQ overflow condition caused the IOMMU to automatically respond to a "Page Request" with the "Last Request in PRG" flag set to 1, then software may observe an incomplete page-request group.  Software should ignore the group and not service such groups.

How is software to know that the IOMMU automatically responded to a Page Request due to a PQ overflow?  Is the intention that, because the IOMMU *might have* automatically responded, if software sees a PQ overflow, then all queued page requests in a group should be discarded when the "Last in PRG" for that group is not already present in the queue?

ved-rivos, 2023-02-07:
Yes, that is correct.

ved-rivos, 2023-02-08:
It could be reworded a bit:
If a `PQ` overflow condition occurred, then software may observe an incomplete page-request group. Software should ignore the group and not service such groups. 

The fact that IOMMU auto-responds is orthogonal to this statement. The auto-response allows recovery from such condition as a success response will cause the device to retry the ATS request leading to another page fault and redoing the page-requests which hopefully now can land in the `PQ` if software has corrected the overflow condition.

ved-rivos, 2023-02-08:
PR #201 updates the guidance

closed 2023-02-09 by ved-rivos