riscv-non-isa/riscv-iommu #206: Execute accesses to IOVA translated through MSI page tables

ved-rivos, 2023-02-11:
Issue #192 introduces a new feature to support MSI conversion even if second stage is Bare. The PR #193 introduced the new feature 

However PR #193 missed the fact that unlike a truly Bare second stage that would provide a SPA=GPA translation with all permissions granted, the MSI page tables do not grant execute permission. 

Before PR #193 transaction that invoked the process to translate address of MSIs causes an "Instruction guest page fault" (cause = 20) for transaction types Untranslated read-for-execute or Translated read-for-execute. However if the second stage is Bare then reporting a guest page fault would not be correct.

One option would be for fault reported to be conditional based on the second stage being Bare i.e. if the second stage is Bare then the fault should be an "Instruction page fault" (cause = 12) and if second stage is not Bare then the fault could be "Instruction guest page fault" (cause = 20). However this would not be correct in case where first stage itself is Bare. 

Proposed fix is to treat interrupt files as being located in a PMA that do not allow instruction execution. This will lead to this condition being treated an access fault and be reported as "Instruction access fault" (cause=1).

ved-rivos, 2023-02-11:
@jhauser-us - please see if the proposed fix is acceptable for AIA.

ved-rivos, 2023-02-13:
PR #207 to address the issue.

jhauser-us, 2023-02-13:
Looks to me like a good resolution.

dgptha, 2023-02-13:
The supposition and fix looks fine. Wonder if it could be moved to an earlier position in the translation process.
Small typo in the line before the fix, "a" to "an": "If the transaction is a Untranslated or Translated read-for-execute..." => "If the transaction is **an** Untranslated or Translated read-for-execute.." or change the order to "Translated or Untranslated"

ved-rivos, 2023-02-13:
> Wonder if it could be moved to an earlier position in the translation process.
Moving to an earlier position may not be desired as this fault wants to be lower priority than any faults caused by an invalid or mis-configured MSI PTE. The process for the two forms of MSI PTEs - basic translate and MRIF mode - converges at this line. 

> Small typo in the line before the fix, "a" to "an"
Thanks! The PR is updated to fix it.

closed 2023-02-15 by ved-rivos