riscv-non-isa/riscv-iommu #229: Process to translate an IOVA

cuihy111, 2023-05-16:
In the process of IOVA address translation, judge whether the signal T2GPA is equal to 1, if it is 1, then IOVA is GPA, carry out the second stage address translation, jump to the 17th step, however, the 17th step says that the first stage address translation problems will stop and report errors, is there some contradiction? The first-stage address translation is not executed. Why is Step 17 used to determine whether The first-stage address translation is incorrect?

ved-rivos, 2023-05-16:
In this case `iosatp.MODE` is `Bare` and there is no first stage address translation or protection.

cuihy111, 2023-05-16:
In Chapter 2.3, step 9 states that iosatp.MODE =Bare when T2GPA=1, jump to step 17. step 17 states that if the first stage address translation encounters problems and stops with an error.why should we check the first stage address translation when iosatp.MODE =Bare.

closed 2023-05-17 by cuihy111

cuihy111, 2023-05-17:
I see in the code that req is translated, untranslated and ATS, but I don't see the specific description of the request type in the document. In Section 2.3, it reads in steps 8 and 9, if it is a translation request Go to steps 8 and 9 to check T2GPA signal, when is the request a translation request and when is it not?