riscv-non-isa/riscv-iommu #238: Can A in the first step of MSI translation process be directly IOVA?

18772820305, 2023-07-17:
Step 1 in the process to translate addresses of MSI in section 2.3.3 let A be the GPA, is the GPA mentioned in it directly from IOVA?

ved-rivos, 2023-07-17:
The process in section 2.3.3 is invoked by the step 18 of the process in section 2.3 and the GPA may be directly from the IOVA or could be the result of the translation using the first stage address translation process.

18772820305, 2023-07-18:
After querying the DC to get msi_addr_mask and msi_addr_pattern, can i directly judge whether the virtual interrupt file is accessed by IOVA through the formula provided in Section 2.1.3.6?

ved-rivos, 2023-07-18:
Section 2.3 outlines the process to translate an IOVA. In step 6, the `device_id` is used to locate the `DC`. At this step whether the IOVA is a GPA is not known. Subsequently the step 17 determines the GPA. Only after step 17 i.e. after a GPA has been determined, the step 18 - MSI address translation process - is invoked. In MSI address translation process, the `msi_addr_mask` and `msi_addr_pattern` from the `DC` determined in step 6 is used - in step 3 - to determine if GPA is that of a virtual interrupt file using the algorithm specified in section 2.1.3.6.

18772820305, 2023-07-19:
Does the address sent by the device for writing to the virtual interrupt file must be a GPA? If it is GPA, can it directly recognize that IOVA is writing virtual interrupt files after obtaining DC.

ved-rivos, 2023-07-19:
The process is then the section 2.3 is the only defined process. 

The address provided by a device is an IOVA. The RISC-V IOMMU employs a two-stage address translation process to translate the IOVA to an SPA. Although there is no option to disable two-stage address translation, either stage may be effectively disabled by configuring the virtual memory scheme for that stage to be Bare i.e. perform no address translation or memory protection. 

So the process following is always:
1. Locate device context - this provides the second-stage address translation controls. The second stage may be Bare
2. If process_id is valid locate the process context - this provides the first-stage address translation controls. The first stage may be Bare
3. Translate  IOVA to a GPA using the first stage - there may not be a translation needed if first stage is Bare
4. Use this GPA to do MSI address translation if it is address of a virtual interrupt file
5. If it is not address of virtual interrupt file translate the GPA to SPA using the second stage - there may not be a translation needed if second stage is Bare

If your questions are about whether in your implementation you can take certain shortcuts or deviate from this process then I can help you if you would more clearly state what you are attempting to do and what your assumptions are.

ved-rivos, 2023-07-21:
Please re-open if there are further items to discuss.

closed 2023-07-21 by ved-rivos

yanhe234, 2024-10-24:
> 那么，第 2.3 节中的过程是唯一定义的过程。合规实施必须遵守此过程。
> 
> 设备提供的地址是 IOVA。RISC-V IOMMU 采用两阶段地址转换过程将 IOVA 转换为 SPA。虽然没有禁用两阶段地址转换的选项，但可以通过将该阶段的虚拟内存方案配置为裸机（即不执行地址转换或内存保护）来有效禁用任一阶段。
> 
> 因此，以下过程始终是：
> 
> 1. 定位设备上下文 - 这提供了第二阶段的地址转换控制。第二阶段可能是裸
> 2. 如果 process_id 有效，则定位进程上下文 - 这提供了第一阶段的地址转换控制。第一阶段可能是裸的
> 3. 使用第一阶段将 IOVA 转换为 GPA - 如果第一阶段是裸阶段，则可能不需要翻译
> 4. 如果它是虚拟中断文件的地址，则使用此 GPA 进行 MSI 地址转换
> 5. 如果不是虚拟中断文件的地址，则使用第二阶段将 GPA 转换为 SPA - 如果第二阶段是裸的，则可能不需要转换
> 
> 如果您的问题是关于在实施过程中是否可以采取某些捷径或偏离此过程，那么如果您能更清楚地说明您正在尝试做什么以及您的假设是什么，我可以为您提供帮助。

Sir, so my understanding is that when translating an MSI address, IOMMU does not know that it is an MSI address. If there is no bare bar in the first stage, then it is necessary to go through one stage of translation to determine whether it is an MSI translation. Is that correct? If so, what are the benefits of this? Why can't we directly determine if it's MSI?

yanhe234, 2024-10-24:
> The process is then the section 2.3 is the only defined process. A compliant implementation must adhere to this process.
> 
> The address provided by a device is an IOVA. The RISC-V IOMMU employs a two-stage address translation process to translate the IOVA to an SPA. Although there is no option to disable two-stage address translation, either stage may be effectively disabled by configuring the virtual memory scheme for that stage to be Bare i.e. perform no address translation or memory protection.
> 
> So the process following is always:
> 
> 1. Locate device context - this provides the second-stage address translation controls. The second stage may be Bare
> 2. If process_id is valid locate the process context - this provides the first-stage address translation controls. The first stage may be Bare
> 3. Translate  IOVA to a GPA using the first stage - there may not be a translation needed if first stage is Bare
> 4. Use this GPA to do MSI address translation if it is address of a virtual interrupt file
> 5. If it is not address of virtual interrupt file translate the GPA to SPA using the second stage - there may not be a translation needed if second stage is Bare
> 
> If your questions are about whether in your implementation you can take certain shortcuts or deviate from this process then I can help you if you would more clearly state what you are attempting to do and what your assumptions are.

Sir, so my understanding is that when translating an MSI address, IOMMU does not know that it is an MSI address. If there is no bare bar in the first stage, then it is necessary to go through one stage of translation to determine whether it is an MSI translation. Is that correct? If so, what are the benefits of this? Why can't we directly determine if it's MSI?

ved-rivos, 2024-10-26:
> Sir, so my understanding is that when translating an MSI address, IOMMU does not know that it is an MSI address. If there is no bare bar in the first stage, then it is necessary to go through one [...]

What the device provides an IOVA. The IOVA has to be first translated to a GPA to determine whether the GPA matches the address of a virtual interrupt file. We cannot directly determine if its a MSI since we do not know the GPA that corresponds to the IOVA without first translating the IOVA.

18772820305, 2025-01-22:
Whether the IOMMU needs to use the MSI page table to translate the MSI sent by the PCIe device in a non-virtualized environment?

ved-rivos, 2025-01-22:
If configured to do so then yes