riscv-non-isa/riscv-iommu #240: Question about iohpmctr in iommu_ref_model

baimengwei, 2023-07-21:
In the spec, iohpmctr is a counter, which seems to be 32-64 bit adjustable by hardware, but if it's 64 bit width for iohpmctr, the C model is out of this range and doesn't seem to report overflow. Is a bug or correct behavior ？

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_hpm.c#L84

if g_reg_file.iohpmctr[i].counter is 0xFFFFFFFFFF，and its width is 40, then a interrupt might be report.

if g_reg_file.iohpmctr[i].counter is 0xFFFFFFFFFFFFFFFF，and its width is 64, then `count` will be zero, a interrupt might be report.

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_hpm.c#L86C14-L86C20

ved-rivos, 2023-07-21:
Yes. Was a bug. Its fixed in PR #241. Thanks for reporting.

baimengwei, 2023-07-22:
Thanks for resolving.

closed 2023-07-22 by baimengwei