riscv-non-isa/riscv-iommu #242: INVAL_DDT command seems to have no PV position, but how to specify PV=0?

baimengwei, 2023-07-24:
[image]

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_sw_guidelines.adoc?plain=1#L147

the iommu spec have the following sentence:

If `DC.tc.PDTV==1` then `IODIR.INVAL_PDT` with `DV=1`, `PV=0`, and `DID=D`

The DV position can be found, but the PV does not.

ved-rivos, 2023-07-24:
That bullet is stale and should be removed. The `IODIR.INVAL_DDT` is sufficient - If `DV` is 1, then the command invalidates cached leaf-level DDT entry for the device identified by `DID` operand and all associated PDT entries. We removed the `PV` operand in `INVAL_PDT` but missed removing this line. Will queue that for an errata update.

ved-rivos, 2023-07-24:
Tracked in PR #243

closed 2023-07-24 by baimengwei