riscv-non-isa/riscv-iommu #245: In which privileged state is it reasonable to execute the IOTINVAL.VMA instruction?

baimengwei, 2023-07-26:
The IOMMU translation-table cache invalidation commands, IOTINVAL.VMA and IOTINVAL.GVMA synchronize updates to in-memory first-stage and second-stage page table data structures respectively with the operation of the IOMMU and invalidate the matching IOATC entries.

However, the Table 9. IOTINVAL .VMA operands and operations contains GV bit, which make it possible for invalidation all the entries including the second-stage page tables. 

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_in_memory_queues.adoc?plain=1#L191

Regarding the IOTINVAL.VMA instruction in the IOMMU spec, in which state is it generally executed. If it is executed in the VS mode, is this command too high-privileged, and all page tables can be wiped out? 

Once a Virtual Machine in VS mode have the permission to wiped out second-stage page tables which used by other Virtual Machine,  the execution of other virtual machines will always be very slow because of the Page Fault, will this be the case?

ved-rivos, 2023-07-26:
These are not instructions. These are commands queued to the IOMMU through a command queue. In a virtualized system, the IOMMU ownership is retained by the hypervisor.

closed 2023-07-27 by baimengwei

18772820305, 2023-08-03:
Can the CPU access all Memory map registers of the IOMMU in all privileged states?

ved-rivos, 2023-08-03:
The IOMMU registers are memory mapped. Like any other memory, the access to that memory depends on the configurations of the PMP and the if enable the virtual memory system. The M mode may make the registers accessible to S mode using appropriate configurations of the PMP registers mapping this address range. The S-mode may for example map them in the page tables with U=1 to allow user mode to access them. So nothing special about IOMMU memory mapped registers compared to any other devices memory mapped registers as far as access control is concerned.

18772820305, 2023-08-03:
Does this mean that if the appropriate PMP and page table permissions are configured, you can also directly access the Memory map registers of the IOMMU in the virtualization mode?

ved-rivos, 2023-08-04:
yes. Whether memory can be accessed by a program depends on the PMP, second-stage and/or first-stage page tables governing memory accesses for that program. If appropriate permissions are provided then the program can access the memory.