riscv-non-isa/riscv-iommu #246: Can the IOMMU serve as a part of an EP device to provide services for the device itself, or can the IOMMU serve as an EP function for other EP devices?

baimengwei, 2023-07-27:
In Chapter 7, there are the following sentences:
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_hw_guidelines.adoc?plain=1#L8
```
The IOMMU may be constructed as a PCIe device itself and be discoverable as a dedicated PCIe
function with PCIe defined Base Class 08h, Sub-Class 06h, and Programming Interface 00h
```

This sentence seems to mean that the IOMMU can be an EP device and serve other EPs. 

Traditionally, the request sent by the device passes through the IO bridge, and the IO bridge operates the IOMMU to complete the address conversion, and sends the converted result to the interconnection bus. The address routing sent by the device is easy to understand, the device request is passed through the IO bridge and IOMMU.

[image]

**What we don't understand is here:**

If the IOMMU serve as an EP function for other EP devices, however, when a EP serves as an IOMMU for other EPs, there is no reason for other EPs to perform address translation through the IOMMU from the hardware connection.

If the IOMMU serve as a part of an EP device to provide services for the device itself, however, the device can use the IOMMU itself, and does not need to provide a function to inform the outside world that there is such a function，which has IOMMU.

Could you please give me some explanation? thank you very much.

ved-rivos, 2023-07-27:
It says the IOMMU may be a PCIe device - similar to how a IO-bridge/host-bridge is a PCI-PCI bridge device. The operations of the IOMMU are as documented in the specification and shown in the picture above. The IOMMU when built as a PCIe function is generic system peripheral with class code as defined in PCIe specification - see [section 1.9](https://pcisig.com/sites/default/files/files/PCI_Code-ID_r_1_11__v24_Jan_2019.pdf)

baimengwei, 2023-07-28:
It is reasonable to say that this IOMMU is used to process the translation request of the EP mounted on the Secondary PCI BUS. Thank you very much.

closed 2023-07-28 by baimengwei