riscv-non-isa/riscv-iommu #247: Why reports a "Transaction type disallowed" fault when the IOVA is determined to be that of a virtual interrupt file (Section 2.1.3.6) and the corresponding MSI PTE is in MRIF mode?

baimengwei, 2023-07-28:
In Chapter 4, there are the following sentences:
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_debug.adoc?plain=1#L13-L15
```
If the IOVA is determined to be that of a virtual interrupt file (Section 2.1.3.6) and the corresponding MSI PTE
 is in MRIF mode, then the process stops and reports a "Transaction type disallowed" (cause = 260) fault.
```

In Chapter 2.6, there are the following sentences:
https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_data_structures.adoc?plain=1#L1262-L1264
```
If the translation request has an address determined to be an MSI address using the rules defined
by the Section 2.1.3.6 but the MSI PTE is configured in MRIF mode then a Success response is
generated with R, W, and U bit set to 1.
```

These two sentences seem to be contradictory, and why `reports a "Transaction type disallowed" fault when the IOVA is determined to be that of a virtual interrupt file and the corresponding MSI PTE is in MRIF mode` ?

If it should `report a "Transaction type disallowed" fault when the IOVA is determined to be that of a virtual interrupt file and the corresponding MSI PTE is in MRIF mode`, this condition might be part of the translation check in the chapter 2.3.3 which named "Process to translate addresses of MSIs".

ved-rivos, 2023-07-28:
The debug interface is modeled similar to the ATS translation request interface. There is only a 64-bit register to return one address translation. MRIF mode translations need to return - notice interrupt ID, address of the notice MSI destination, and the address of the destination MRIF. That is why the debug interface does not support it.

baimengwei, 2023-07-28:
The explanation is very clear, thank you.

baimengwei, 2024-11-15:
Hi. sir

is the c model checks the msi pte in mrif mode which should report fault ? I can not find the matched code with the spec. 

https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_ref_model/libiommu/src/iommu_reg.c#L706

ved-rivos, 2024-11-16:
Thanks for spotting that. That check was missing. Its now added by #447

closed 2024-11-16 by ved-rivos