riscv-non-isa/riscv-iommu #249: Is there a situation where DevATC stores SPA and the device is controlled by Guest OS?

baimengwei, 2023-08-04:
Under the control of the Guest OS, when the device initiates a VA or GPA address translation request, then the IOMMU determines that the address is related to the MSI, so IOMMU searches and returns the translation result. 

According to the [Process to translate addresses of MSIs](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_data_structures.adoc#MSI_TRANS), Which address will be returned to DevATC, is GPA or SPA ?

Or, Is there any SPA address that the Guest OS can perceive from address translation request and cached SPA in DevATC? 

If the Guest OS can obtain any information about the SPA address, can it be guaranteed that no security issues will be caused?

ved-rivos, 2023-08-04:
The deviceATC receives a `GPA` if the `T2GPA` control is active otherwise it receives a `SPA`. Typically use of ATS involves providing `SPA` to  the device. Whether the device makes the `SPA` in the devATC available to software depends on the device. Typically devices do not do that but I cannot make a statement about all devices. Whether a device is secure and can be trusted is a determination that the platform administrator has to make. The IOMMU cannot provided any guarantees or make claims about the security of a device.

baimengwei, 2023-08-07:
> Whether a device is secure and can be trusted is a determination that the platform administrator has to make. The IOMMU cannot provided any guarantees or make claims about the security of a device.

Thanks.

closed 2023-08-07 by baimengwei