riscv-non-isa/riscv-iommu #252: Could you please explain why it is not necessary to set V bit to 0 when doing a page size promotion or demotion?

baimengwei, 2023-08-23:
In most cases abort mmu, it seems that the update of the page table adopts the BBM(Break Before Make) mechanism to ensure that other harts get the correct new page table. This rule seems to be applicable to iommu.

[the spec](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_sw_guidelines.adoc#accessed-adirty-d-bit-updates-and-page-promotions) has the sentence below:
```
The only PTE update supported by the IOMMU without first clearing the V bit in the original PTE 
and executing a appropriate IOTINVAL command is to do a page size promotion or demotion. 
```

[the spec](https://github.com/riscv-non-isa/riscv-iommu/blob/main/iommu_in_memory_queues.adoc#command-queue-cq) has the sentence below:
```
a page is upgraded to a larger page without first clearing the original non-leaf PTE’s valid bit and executing
 an IOTINVAL.VMA or IOTINVAL.GVMA as applicable with AV=0. In this case, a similar remark applies: it is 
unpredictable whether the old non-leaf PTE or the new leaf PTE is used, but the behavior is otherwise well defined.
```

The second reference seems to indicate that the V position needs to be cleared, and then the INVAL instruction shoule be executed, but it seems to conflict with the first reference.

Is there really a conflict between these two references? Could you please explain why it is not necessary to set V bit to 0 when doing a page size promotion or demotion?

baimengwei, 2023-08-23:
Seems like a misunderstanding. The first is the translation mode change, the second is the page table lookup change.

ved-rivos, 2023-08-25:
> page size promotion or demotion

Page size promotion refers to the process where 512 pages where each page is of size X are merged to create a single mapping with size 512 * X. For example, 512 pages mapped as 4 KiB pages may be merged to create a single 2 MiB page. Page size demotion refers to the opposite process.

baimengwei, 2023-08-27:
I see, thank you a lot.

closed 2023-08-27 by baimengwei