riscv-non-isa/riscv-iommu #254: Can you clarify "command-queue access"

hsxyb, 2023-08-28:
Can you clarify what does "command-queue access" mean in description of cqmf ?includecommand queue read access and the write access of 
https://github.com/riscv-non-isa/riscv-iommu/blob/61026efa47a8a4fdcf301788044aba2308add3bf/iommu_registers.adoc?plain=1#L782-L786

ved-rivos, 2023-08-28:
> Does it include command queue read access and the write access of IOFENCE.C command with AV=1?

Yes. It includes both.

ved-rivos, 2023-08-28:
Would be a good clarification to include - see https://github.com/riscv-non-isa/riscv-iommu/pull/243/commits/69026209b298751ad6b3ee1e7ee928262d82b222.

closed 2023-08-29 by hsxyb

wangyongzhen0322, 2024-06-27:
If the write access failure of IOFENCE.C command with AV=1 will set the cqmf to 1, the cqh will reference the IOFENCE.C command. But a IOFENCE.C command completion is determined by cqh advancing past the index of the IOFENCE.C command in the CQ and the IOMMU writes data to memory with AV=1 when the IOFENCE.C command completes. It seems that there is confusion in the description of the spec. When a write access leads a failure with AV=1, wheather the IOFENCE.C is complete?Using only cqh and cqmf does not seem to be enough; it cannot indicate both IOFENCE.C completion and memory failure.

ved-rivos, 2024-06-27:
> When a write access leads a failure with AV=1, wheather the IOFENCE.C is complete?

The IOFENCE.C is not complete because it encounters a memory fault.

wangyongzhen0322, 2024-06-28:
Thanks for your answer. And could you consider revising the sentence in the spec, as it reads very ambiguously in terms of time relationships?
> If AV=1, the IOMMU writes DATA to memory at a 4-byte aligned address ADDR[63:2] * 4 as a 4-byte store when the command completes.

wangyongzhen0322, 2024-06-28:
And in the function do_fence_c, the bit fence_w_ip will be set to 1 no matter whether the write access is successful if the AV=1. So is there an error in the c_model?

ved-rivos, 2024-06-28:
>  And could you consider revising the sentence in the spec, as it reads very ambiguously in terms of time relationships?

The `DATA` is written to the `ADDR` if the command completes but if the write to `ADDR` faults then no data is written i.e. no data is written unless the command completes and completing the command involves writing the data.

ved-rivos, 2024-06-28:
Included a clarification here https://github.com/riscv-non-isa/riscv-iommu/pull/243/commits/67b7b83556d6a87762d42dec2ef59b51b6830dbf